morrownr/8814au #94: raspberry pi kernel / can it adapt older kernels-headers?

FahdSaadA, 2022-12-10:
hi there, 
Previously i had this issue, regarding few limited list of kernels supporting the firmware..
I was wondering..is it possible anyway to use the router with other kernels? sometimes we stock with older headers...

For example i want to use this router with raspberry pi 4b 5.10.92 which is not supported..it supports 5.15 and 5.10.103 ..

with many thanks for this interesting work..

morrownr, 2022-12-11:
Hi @FahdSaadA 

I am not sure if I understand what the problem is. Can you rephrase it?

Nick

FahdSaadA, 2022-12-11:
right now, this Linux distribution for raspberry pi was tested with Raspberry Pi OS  (kernel 5.15) which is the latest Raspian..however, i had tested it with older kernel. it did not work it only works with latest once
but, i want to use it with older version ( instead of 5.15 >i want to use >>>5.10)

how to add this older kernel header to be used?

morrownr, 2022-12-11:
Okay, you are running a RasPi4B with the RasPiOS. You have an sd card with a fairly recent version of RasPiOS that has kernel 5.15. This driver installs and works.

You have another sd card with an older version of RasPiOS that has kernel 5.10 and something is wrong.

Is that correct?

FahdSaadA, 2022-12-11:
that is right..in fact, for 5.10, there are many versions under 5.10, i have tried earlier 5.10.103 earlier and it was working fine..i want to use this tool with 5.10.92 kernel header...
this header is not supported, when follow the setup process, the kernel 5.10.92 was not found...!!
hope you got the issue..
may thanks again

morrownr, 2022-12-12:
>  i have tried earlier 5.10.103 earlier and it was working fine

That tells me that you had kernel 5.10.103 kernel header 5.10.103. The kernel and headers have to be an exact match.

> i want to use this tool with 5.10.92 kernel header...
this header is not supported,

Support for older kernels and the header files that go with them are depreciated depending on the policy of the distro managers. 

I do my best to help users with this driver but there is nothing I can do about this issue. This is an issue for the manager of the distro.

Nick