morrownr/8814au #95: AP mode seems broken on kernel 6.0.0

Palkovsky, 2022-12-17:
I encountered a problem when creating AP on a machine with 6.0.0 kernel. Perhaps the cause lies within hostapd (mssing an option in confg or some dependency on the test machine?), but I suspect the driver as the main culprit.

**Adapter**: Alfa AWUS 1900
**Machine 1**: `Linux ubuntu 5.15.0-50-generic #56~20.04.1-Ubuntu SMP Tue Sep 27 15:51:29 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
**Machine 2**: `Linux kali 6.0.0-kali5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.0.10-2kali1 (2022-12-06) x86_64 GNU/Linux`

**hostapd.conf**:
```
interface=wlan0
ssid=Test-SSID
channel=11

# 802.11n
hw_mode=g
ieee80211n=1

# WPA2 PSK with CCMP
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=12345678
```

**Reproduction**:
Compile and install driver using default options on both machines, then start *hostapd* with provided config.

**Machine 1**:
```
test@ubuntu:/tmp$ sudo hostapd hostapd.conf 
Configuration file: hostapd.conf
Using interface wlan0 with hwaddr 00:c0:ca:b0:7d:84 and ssid "Test-SSID"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
```

Everything works as expected, other devices can see the AP.

**Machine 2**:
```
test@kali:/tmp$ sudo hostapd hostapd.conf
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: INTERFACE-DISABLED 
wlan0: INTERFACE-ENABLED 
nl80211: kernel reports: key not allowed
Failed to set beacon parameters
wlan0: IEEE 802.11 driver starting channel switch: freq=2422, ht=1, vht_ch=0x0, he_ch=0x0, offset=1, width=2 (40 MHz), cf1=2432, cf2=0
wlan0: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2422 ht_enabled=1 ch_offset=1 ch_width=40 MHz cf1=2432 cf2=0 dfs=0
```

Other devices don't see the AP.

Palkovsky, 2022-12-17:
Seems like the issue was some kind of a glitch. I'm not able to reproduce it now and the AP mode works as expected.

closed 2022-12-17 by Palkovsky

morrownr, 2022-12-17:
> Seems like the issue was some kind of a glitch. I'm not able to reproduce it now and the AP mode works as expected.

I've noticed that Kali has been somewhat prone to glitches over the last several months. Well, actually Kali has never been as stable as Ubuntu. If you are setting up an AP, you might want to avoid Kali.