morrownr/8814au #96: 6.0.0-kali5-amd64  [ 0| 0 ACKs] -- no handshake, how to fix?

iDx-a, 2022-12-18:


morrownr, 2022-12-18:
Hi @iDx-a 

If you will provide the information I need to duplicate the result here, I will investigate this as able. I'll need to know what app you are using and what you are doing with the app as a minimum.

Nick

iDx-a, 2022-12-18:
Hi @morrownr ,

First, thanks for fast answer.
Installation process was OK without any error.
also NO  problem to change mac address  and put in monitor mode using ifconfig, iwconfig and airmon-ng, NO problem  in airodump-ng for scan.
aireplay-ng to make deauth and make handshake is problem

For more info printscreen etc... , don`t  hesitate to contact me.

Best Regards,
iDx

morrownr, 2022-12-18:
> aireplay-ng to make deauth and make handshake is problem

Okay, so the app where you are seeing problems is aireplay-ng. I have that installed on some test systems.

For deauth and handshake, can you give me a checklist that shows the exact commands you are using so that I can duplicate the problem?

If I can exactly duplicate the problem, it will make it easier to discover a solution. You can use printscreen along with a written checklist if you want.

Regards

iDx-a, 2022-12-18:
you need print screen from aireplay-ng in terminal ?

morrownr, 2022-12-18:
> you need print screen from aireplay-ng in terminal ?

If that shows me the exact commands you are using and the results, yes. I would prefer that you also type each command you are using. I need to duplicate what you are doing and since I am not there to see what you are doing, you need to show me,

iDx-a, 2022-12-18:
> If that shows me the exact commands you are using and the results, yes. I would prefer that you also type each command you are using. I need to duplicate what you are doing and since I am not there to see what you are doing, you need to show me,
if you have time you can connect on my pc and check?

iDx-a, 2022-12-22:
> > you need print screen from aireplay-ng in terminal ?
> 
> If that shows me the exact commands you are using and the results, yes. I would prefer that you also type each command you are using. I need to duplicate what you are doing and since I am not there to see what you are doing, you need to show me,

Hi @morrownr ,

Soryy bro, but i was busy,

[image]

iDx-a, 2022-12-22:
> > If that shows me the exact commands you are using and the results, yes. I would prefer that you also type each command you are using. I need to duplicate what you are doing and since I am not there to see what you are doing, you need to show me,
> 
> if you have time you can connect on my pc and check?

Hi @morrownr ,

Sorry bro i was busy,

[image]

morrownr, 2022-12-25:
Hi @iDx-a 

Sorry about the delay. Holidays ya know. I should have time to look at this sometime after the first of the year so yell at me then if you have not figured it out.

Nick

liviupopa16, 2023-01-10:
did you manage to fix the problem? i get the same 0 ACKS, but my client gets sometimes disconnected

iDx-a, 2023-01-11:
> did you manage to fix the problem @iDx-a ? i get the same 0 ACKS, but my client gets sometimes disconnected
Hi @liviupopa16 ,
No, i cant disconnect client.
you have same awus 1900?

iDx-a, 2023-01-18:
Hi @morrownr ,

I just want to inform you that problem i solved.
i make reinstall on kali OS and of course and new driver installation,  still [ 0| 0 ACKs]  but client is successfully disconnected and make handshake.

Best Regards,

morrownr, 2023-01-18:
@iDx-a @liviupopa16 

Fixing the 0|0 ACKs problem will likely take a deep dive and I am too busy right now. Check in with me in about a month and maybe I can work on it.

Nick

iDx-a, 2023-01-18:
> @iDx-a @liviupopa16 
> 
> Fixing the 0|0 ACKs problem will likely take a deep dive and I am too busy right now. Check in with me in about a month and maybe I can work on it.
> 
> Nick

Ok Nick

KiGamji, 2023-02-11:
@liviupopa16 @iDx-a, Probably kernel issue. Iirc, it doesn't support correct frame injection by default. I can look into it a bit later, feel free to ping me if I'll not send anything here for a few days.

iDx-a, 2023-02-11:
> @liviupopa16 @iDx-a, That's probably kernel issue. Iirc, it doesn't support correct frame injection by default. I can look into it a bit later, feel free to ping me if I'll not send anything here for a few days.

Thanks @liviupopa16, i will

morrownr, 2023-02-11:
> @liviupopa16 @iDx-a, That's probably kernel issue. Iirc, it doesn't support correct frame injection by default. I can look into it a bit later, feel free to ping me if I'll not send anything here for a few days.

@KiGamji 

I'm looking forward to seeing what you come up with. I'm hoping to have some extra time to work on this driver soon. I have been doing major upgrades to the other 4 Realtek drivers here but this driver is a challenge. It is old code that is so dissimilar to the other drivers that it is hard to work with and maintain. I wish that Realtek would release a modernized version of this driver but it may be that they are not even maintaining this driver anymore.

Nick

KiGamji, 2023-02-13:
@iDx-a @liviupopa16 So, I checked out the code, and it seems, that the packet injection part didn't change. Can you try to apply this patch? https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-kernel/-/blob/master/patches/4.14/add-wifi-injection-4.14.patch
Just go to your kernel source folder, put this file to it and run this command: `patch -p1 < add-wifi-injection-4.14.patch`

iDx-a, 2023-02-13:
> @iDx-a @liviupopa16 So, I checked out the code, and it seems that the packet injection part of it didn't really change since 4.14 (fixing patch was made for 4.14) Can you try to apply this patch? https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-kernel/-/blob/master/patches/4.14/add-wifi-injection-4.14.patch Just go to your kernel source folder, put this file to it and run this command: `patch -p1 < add-wifi-injection-4.14.patch`

hey @KiGamji ,

sorry and thank you, but i`m still n00b. If is not problem for you can  you explain me how to do this, please?

KiGamji, 2023-02-13:
> sorry and thank you, but i`m still n00b. If is not problem for you can you explain me how to do this, please?

 uhhh, so, you need to learn how to build linux kernel, because you will anyways need a kernel with this patch, so packet injection will work. you can find guides in the internet, its not that hard. after you learn about it a bit, you will be easily able to check if this patch applies successfully

KiGamji, 2023-02-13:
@iDx-a btw, I recommend you to use wifite instead of airodump-ng and airmon-ng. Just read about what it does

iDx-a, 2023-02-13:
> uhhh, so, you need to learn how to build linux kernel, because you will anyways need a kernel with this patch, so packet injection will work. you can find guides in the internet, its not that hard. [...]

Ok, thank you, i will try and write here later

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided) (reporter did not include reproducible details; request full healthcheck and command transcript. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.