morrownr/8814au #97: Issue while installing on nethunter

cr4sh-me, 2023-01-19:
Hi, I was trying to install driver on a nethunter device.

Here's output:

[image]

morrownr, 2023-01-19:
Hi @cr4sh-me 

I am not familiar with Nethunter but if you want to provide a link so that I can read about it as I have time, I'll try to do that.

I see that kernel 3.18 is in use. The oldest kernel supported by the Linux foundation is currently 4.14 and I generally stick to supporting supported kernels. I don't have a single test system that has a kernel that old on it so my ability to work this issue is zero.

Something that might help you: The Main Menu is located at:

https://github.com/morrownr/USB-WiFi

It has a lot of information about usb wifi adapter and you can ask question in `Issues.` That might help you find a solution.

Regards,

Nick

cr4sh-me, 2023-01-27:
https://www.kali.org/docs/nethunter/

Okay, ill try

morrownr, 2023-01-27:
https://www.kali.org/docs/nethunter/wireless-cards/

It sounds to me that the driver support is already provided in the Nethunter kernel.

After reading, it looks like a 8814au chipset based adapter is not really going to work well. These adapter can take up to 900 mA of power. Something with a mt7610u chipset that does not use much power might be better suited. An adapter like the Alfa ACHM could work well but you would have to research the issue.

cr4sh-me, 2023-01-27:
> https://www.kali.org/docs/nethunter/wireless-cards/
>
> It sounds to me that the driver support is already provided in the Nethunter kernel.
>
> After reading, it looks like a 8814au chipset based adapter [...]

I had Nexus 6P and this card worked fine with no additionals drivers and with OTG cable only, but the kernel had 8814au. Official OnePlus3 nethunter kernel its not working, so i need to compile my own.

morrownr, 2023-01-27:
> so i need to compile my own.

The other drivers I have up here are pretty good quality. This one, not so much. It is old and fragile and Realtek seems to have stopped releasing code for it. I would help but you are trying to do something that is outside of my expertise.

KiGamji, 2023-02-13:
@cr4sh-me since the issue is solved, can you close it as completed?

morrownr, 2023-02-14:
Sure.

closed 2023-02-14 by morrownr