morrownr/8814au #100: Installation problem arises in ioctl_cfg80211.c

tomkark, 2023-03-20:
Hey,
I've clean installed Ubuntu 22.04 today on my new SSD and since i use an AC1900 USB 3.0 WiFi Adapter (A7000) i needed to install the 8814au drivers, although when i tried to do it, it failed, even tried downgrading the kernel twice and it didn't help, when i ran `make` manually the problem that prevented my build from finishing was  on line 451 in `os_dep/linux/ioctl_cfg80211.c`:
```
if (!rtw_cfg80211_allow_ch_switch_notify(adapter))
	goto exit;
```
an implicit declaration of `rtw_cfg80211_allow_ch_switch_notify`, i tried to find it's definition but i failed, had some research online but no one seemed to talk about it ever. i also tried adding a definition manually in the headers which didn't work as well, finally after some time i commented out these 2 lines wondering if anything would change, and for my surprise the build passed and now my adapter works.

My question is, why did this happen to me and how damaging is the commenting of these 2 lines?

I'm seeing this as a temporary solution until i resolve it.

Current kernel: 5.19.0-35-generic

fa1ke5, 2023-03-20:
Аналогичная проблема
wifi/8814au/os_dep/linux/ioctl_cfg80211.c:451:14: error: implicit declaration of function ‘rtw_cfg80211_allow_ch_switch_notify’; did you mean ‘rtw_cfg80211_ch_switch_notify’? [-Werror=implicit-function-declaration]
  451 |         if (!rtw_cfg80211_allow_ch_switch_notify(adapter))
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |              rtw_cfg80211_ch_switch_notify

tomkark, 2023-03-20:
> 

Yes, this is exactly what I meant. +

tomkark, 2023-03-20:
> Аналогичная проблема wifi/8814au/os_dep/linux/ioctl_cfg80211.c:451:14: error: implicit declaration of function ‘rtw_cfg80211_allow_ch_switch_notify’; did you mean ‘rtw_cfg80211_ch_switch_notify’? [...]

Yes, this is exactly what I meant. +

morrownr, 2023-03-20:
It is possible that I broke something. Let me see about it.

tomkark, 2023-03-20:
> It is possible that I broke something. Let me see about it.

I did it a couple of hours ago so I may me mistaken, but IIRC, I also tried going back to 752d8ea365b2affc5d356e35659600995508849d
and it failed on the same thing as well, but don't rely on my word, my memory is burnt out

morrownr, 2023-03-20:
Okay, the error is coming from code I worked on and merged yesterday. The old code in this driver can be difficult to work on.

morrownr, 2023-03-20:
@tomkark 
@fa1ke5 

It is a long story what I did but it was my fault. I did a merge that depended on another merge that I forgot to merge.

Anyway, I just merged a lot of stuff. The docs, scripts and even some code has been midifed so please test and let me know how it goes. You may was to clean things out first, then update and then install:

sudo sh remove-driver.sh
git pull
sudo sh install-driver.sh

morrownr, 2023-03-21:
@tomkark
@fa1ke5

I've merged more. Things seem to be working again but you may need to reread the docs as many upgrades to the docs and scripts have been merged. This is especially true if not running on x86_64 as the installations script now auto detects the platform and handles things accordingly.

tomkark, 2023-03-21:
> @tomkark @fa1ke5
>
> I've merged more. Things seem to be working again but you may need to reread the docs as many upgrades to the docs and scripts have been merged. This is especially true if not [...]

Worked!

Thank you very much.

closed 2023-03-21 by tomkark