morrownr/8814au #103: array-index-out-of-bounds during resume from hibernate

user-8472, 2023-04-02:
Observed a crash of driver during a resume from hibernate. The system was able to recover gracefully. Call Traces are attached.   The designated (-->) lines below show the offending lines.

```
switch (rf_path) {                                                                                                                      
case RF_PATH_A:                                                                                                                         
    get_mix_mode_tx_agc_bb_swing_offset(dm, method, (enum rf_path)rf_path, tx_power_index_offest);                                      
    odm_set_bb_reg(dm, REG_A_TX_AGC, TXAGC_BITMASK, cali_info->absolute_ofdm_swing_idx[rf_path]);                                       
-->odm_set_bb_reg(dm, REG_A_BBSWING, BBSWING_BITMASK, tx_scaling_table_jaguar[cali_info->bb_swing_idx_ofdm[rf_path]]); /*set BBswing*/ 
    break;                                                                                                                              
                                                                                                                                        
case RF_PATH_B:                                                                                                                         
    get_mix_mode_tx_agc_bb_swing_offset(dm, method, (enum rf_path)rf_path, tx_power_index_offest);                                      
    odm_set_bb_reg(dm, REG_B_TX_AGC, TXAGC_BITMASK, cali_info->absolute_ofdm_swing_idx[rf_path]);                                       
-->odm_set_bb_reg(dm, REG_B_BBSWING, BBSWING_BITMASK, tx_scaling_table_jaguar[cali_info->bb_swing_idx_ofdm[rf_path]]); /*set BBswing*/ 
    break;                                                                                                                              
                                                                                                                                        
case RF_PATH_C:                                                                                                                         
    get_mix_mode_tx_agc_bb_swing_offset(dm, method, (enum rf_path)rf_path, tx_power_index_offest);                                      
    odm_set_bb_reg(dm, REG_C_TX_AGC, TXAGC_BITMASK, cali_info->absolute_ofdm_swing_idx[rf_path]);                                       
-->odm_set_bb_reg(dm, REG_C_BBSWING, BBSWING_BITMASK, tx_scaling_table_jaguar[cali_info->bb_swing_idx_ofdm[rf_path]]); /*set BBswing*/ 
    break;                                                                                                                              
                                                                                                                                        
case RF_PATH_D:                                                                                                                         
    get_mix_mode_tx_agc_bb_swing_offset(dm, method, (enum rf_path)rf_path, tx_power_index_offest);                                      
    odm_set_bb_reg(dm, REG_D_TX_AGC, TXAGC_BITMASK, cali_info->absolute_ofdm_swing_idx[rf_path]);                                       
-->odm_set_bb_reg(dm, REG_D_BBSWING, BBSWING_BITMASK, tx_scaling_table_jaguar[cali_info->bb_swing_idx_ofdm[rf_path]]); /*set BBswing*/ 
    break;                                                                                                                              
                                                                                                                                        
default:                                                                                                                                
    RF_DBG(dm, DBG_RF_TX_PWR_TRACK,                                                                                                     
           "Wrong path name!!!!\n");                                                                                                    
    break;                                                                                                                              
}                                                                                                                                       
```

[array-index-out-of-bounds.8814.txt](https://github.com/morrownr/8814au/files/11130951/array-index-out-of-bounds.8814.txt)

morrownr, 2023-04-03:
Hi @user-8472 

I'm too busy to take a look this week but maybe next week. This is a subject where I don't have a lot of knowledge and I don't test the drivers here with hibernation. Maybe someone else with more knowledge will come along.

@morrownr

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: clamp `bb_swing_idx_ofdm` before indexing `tx_scaling_table_jaguar` in rtl8814a power-tracking paths. This included a direct code-path fix and validation in the current maintenance cycle.