morrownr/8814au #106: TP-LINK TL-WDN7200H Wireless Adapter Connectivity Issues on Arch Linux with Kernel 6.3.1 Zen

tategotoazarasi, 2023-05-04:
Hello @morrownr and contributors,

First of all, I'd like to thank you for your work on the RTL8814AU driver. I am using the TP-LINK TL-WDN7200H, an external USB wireless adapter, on Arch Linux with the recently upgraded Linux kernel 6.3.1 Zen. Unfortunately, I encountered connectivity problems after the kernel upgrade.

Here is a brief summary of the problem:

1. The LED on the adapter does not light up.
2. The adapter will not automatically connect to Wi-Fi.
3. When I try to manually connect to Wi-Fi using NetworkManager, it gets stuck at "Setting the network address". The network is accessible during this time, but after a short time the connection fails and I get a message saying "IP address unavailable".

I tried the blackarch/rtl8814au-aircrack-dkms package and the rtl8814au-dkms-git package from AUR, but both resulted in installation errors. I then tried the latest version of the driver from this repository, which installed without errors. The network was accessible after a reboot, but the connection eventually failed after some time, just like before.

My setup:

- OS: Arch Linux
- Kernel: 6.3.1 Zen
- Network adapter: TP-LINK TL-WDN7200H
- Driver: RTL8814AU (from this repository)

I understand that maintaining this driver can be challenging, but I would appreciate your help in resolving this issue. Please let me know if you need any additional information or if there are any troubleshooting steps you would like me to try. I appreciate any help you can give me.

Thank you very much for your time and help.

ljluestc, 2023-06-25:
check compatibility with Linux kernel 6.3.1 Zen?

morrownr, 2023-06-25:
Hi @tategotoazarasi 

> I encountered connectivity problems after the kernel upgrade.

Did you upgrade the driver on your system before upgrading the kernel? If not:

$ git pull
$ sudo sh install-driver.sh

Let me know that works.