morrownr/8814au #109: Could not build on kernel 5.15

qiujiangkun, 2023-05-25:
```log
                               jack@jack-itx
██████████████████  ████████   -------------
██████████████████  ████████   OS: Manjaro Linux 23.0.0 UltimaThule x86_64
██████████████████  ████████   Kernel: 5.15.113-1-MANJARO
██████████████████  ████████   Uptime: 15 mins
████████            ████████   Packages: 1722 (pacman), 11 (cargo)
████████  ████████  ████████   Shell: zsh 5.9
████████  ████████  ████████   Terminal: /dev/pts/0
████████  ████████  ████████   CPU: AMD Ryzen 7 5700G (16) @ 3.800GHz
████████  ████████  ████████   GPU: NVIDIA GeForce GTX 1660
████████  ████████  ████████   Memory: 1048.43 MiB / 31966.42 MiB
████████  ████████  ████████
████████  ████████  ████████
████████  ████████  ████████
████████  ████████  ████████

➜  ~ sudo dkms install --no-depmod rtl8814au/5.8.5.1.r161.g6f80699 -k 5.15.113-1-MANJARO
DKMS make.log for rtl8814au-5.8.5.1.r161.g6f80699 for kernel 5.15.113-1-MANJARO (x86_64)
Fri May 26 12:25:52 AM HKT 2023
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.113-1-MANJARO/build M=/var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build  modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 12.2.1 20230201
  You are using:           gcc (GCC) 13.1.1 20230429
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_ioctl_query.o
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_ieee80211.o
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_cmd.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_security.o] Error 1
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_io.o] Error 1
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_debug.o] Error 1
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_mlme.o
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_ioctl_query.o] Error 1
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_ioctl_set.o] Error 1
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_ieee80211.o] Error 1
cc1: error: cannot load plugin ./scripts/gcc-plugins/structleak_plugin.so: ./scripts/gcc-plugins/structleak_plugin.so: undefined symbol: tree_code_type
make[2]: *** [scripts/Makefile.build:289: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build/core/rtw_mlme.o] Error 1
make[1]: *** [Makefile:1905: /var/lib/dkms/rtl8814au/5.8.5.1.r161.g6f80699/build] Error 2
make: *** [Makefile:2393: modules] Error 2
```

morrownr, 2023-05-26:
Hi @qiujiangkun 

> sudo dkms install --no-depmod rtl8814au/5.8.5.1.r161.g6f80699 -k 5.15.113-1-MANJARO

I realize there are multiple ways to compile and install the driver but the method I support is the one spelled out in the Installation Steps in the README. Recommend you clean up and follow the Installation Steps.

One last word. I see you are using gcc 13. I test with and support gcc 9 through 12 but I am not there yet with 13.

Regards

qiujiangkun, 2023-05-26:
Thanks. I'll try the approach in readme and gcc12. The version in AUR is broken after the upgradation though

morrownr, 2023-05-26:
> upgradation 

I certainly try to support Manjaro users but am not that familiar with some of the details of that distro and i am not the one that handles the AUR version. I'm not sure what you mean by "upgradation" so let me know what that means and I'll try to help.

Two more things:

1. I am going on the road in the morning and may not be available for a few days.
2. The reason I try to get users to use the installation steps per README is that the installation script, install-driver.sh, has a lot of extra code in it to detect and handle errors that may come up and often it can tell us what the problem is right on the screen.

Regards

qiujiangkun, 2023-05-26:
Manjaro and Arch are rolling release version, and they don't care about AUR compatibility. Sometimes upgrading kernel breaks something. I'll try the recommended method and potential fix that AUR package

qiujiangkun, 2023-06-09:
I installed 8814 with recommended method on 5.10, then reinstall/upgrade linux with pacman, "dkms install" are triggered with proper environments and it compiles.
==> dkms install --no-depmod rtl8814au/5.8.5.1 -k 5.15.115-1-MANJARO

Hope this helps someone

closed 2023-06-09 by qiujiangkun