morrownr/8814au #111: When compiling openwrt, add 8814au driver

junx964, 2023-08-07:
I refer to https://github.com/morrownr/88x2bu-20210702/issues/42 and used [8814au.tar.gz](https://github.com/morrownr/88x2bu-20210702/files/8403712/rtl8814au.tar.gz)，But now it cannot be compiled successfully. Is it not applicable with the update of openwrt?

The openwrt source code I use is [coolsnowwolf/lede](https://github.com/coolsnowwolf/lede)

I put the decompressed 8814au in package/kernel/

morrownr, 2023-08-07:
Hi @junx964 

I use OpenWRT and USB WiFi adapters together but none of the adapters that I use with OpenWRT use Realtek chipsets. It is simply too much work, especially when you consider the in-kernel drivers that are already included with OpenWRT.

OpenWRT has drivers for the mt7921au, mt7612u and mt761u chipsets plus many more. Main Menu item 2 has a list of chipsets and adapters that use in-kernel drivers: My personal favorite adapter for use with OpenWRT is the Alfa ACM.

https://github.com/morrownr/USB-WiFi

Hope this helps.

@morrownr

junx964, 2023-08-08:
> 
I use [EW-7833UAC](https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_ac1750_dual-band/ew-7833uac/), the official customer service said to use the 8814au driver, what should I do?

can https://github.com/morrownr/USB-WiFi satisfy me

morrownr, 2023-08-08:
@junx964 

> the official customer service said to use the 8814au driver

I'll estimate that around 98% of USB WiFi adapter retailers don't have adequate knowledge to give advice anyone about Linux. I can tell you a couple that actually do support Linux if you are interested.

> , what should I do?

I answered this in the last post.

> can https://github.com/morrownr/USB-WiFi satisfy me

Highly unlikely, unless you are a skilled programmer. Most Realtek based adapters, including the 8814au chipset, are supported with out-of-kernel drivers, the driver in this repo is known as an out-of-kernel driver. Said drivers are primarily designed to support programmers of embedded products. These drivers are not designed for basic desktop systems like Ubuntu, Debian, Raspberry Pi OS, Manjaro and other mainstream desktops. I have done a lot to make these drivers easier and more dependable for regular desktop users but even with the work I have done, I cannot recommend users buy some chipsets and the 8814au is one of those. Sure I have an 8814au based adapter and I make use of it for specific things but I use my experience to try to show users what is best for them based on their use case.

With that said, you are trying to make your 8814au based adapter work on a distro that is VERY specialized. It has to be specialized as it has to work on routers that have very low system resources. You can't just downloaded and compile this driver on OpenWRT. Do you know how to do cross-compiling? If not, you can learn but how much time do you have to invest in making this work. 

The USB-WiFi repo, which is where the Main Menu is located here, gets around 14,000 hits per week. I can see where the users are coming from. OpenWRT is 7th on the list. The OpenWRT users are not coming to USB-WiFi to find out how to compile Realtek drivers, they are coming to find the good adapters where they don't have to compile anything.

Again, I recommend you go to:

https://github.com/morrownr/USB-WiFi

That is the site Main Menu. Recommend you read menu items 1, 2 and 4.

As your knowledge grows, you can ask questions in Issues at USB-WiFi.

I will be glad to advise you if you need help picking an appropriate adapter.

Regards,

@morrownr

junx964, 2023-08-08:
> @junx964
>
>
> I'll estimate that around 98% of USB WiFi adapter retailers don't have adequate knowledge to give advice anyone about Linux. I can tell you a couple that actually do support Linux if you [...]

Thank you, I see why it's not supported anymore. Too much regret, I bought this device without doing enough homework！

morrownr, 2023-08-10:
Hi @junx964 

> Too much regret, I bought this device without doing enough homework！

I understand. You are not alone. The primary reason I started USB-WiFi was to help Linux users with information regarding what USB WiFi adapter to buy for their use case. There is nobody that helps with USB-WiFi that makes any money from their effort. It is all about getting good information to Linux users. I hope to see you asking questions over at USB-WiFi.

@morrownr

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.