morrownr/8814au #112: 5ghz network not visible

ohno1052, 2023-08-17:
Hello! Installed arch linux, your driver, and my 5ghz network is not avaliable. But some other 5ghz nets around are accessible, so, i guess, its smth with my router settings, I dont know much about networks and hope you will be able to discover which settings I need to change to be able to connect to my wifi.
[image]

ohno1052, 2023-08-17:
on win 11 everything is fine tho

morrownr, 2023-08-17:
Hi @ohno1052 

I'll do my best.

> so, i guess, its smth with my router settings,

This very well can be fixed with a changed router setting but I would not blame the router until the router is found guilty in a court of law.  Ha ha.

What could be the problem is that this driver may not work well with a specific router setting so if we can find that setting and it still works well in Windows, life is good.

FYI: This specific driver, as made available from Realtek, was NEVER a good driver. It is missing many features and work on it ended in 2019 so it is also an old driver. I do my best to keep it going but it is a hard driver to work on as it is somewhat fragile.

As far as specific settings are concerned: Trial and error is the best I can offer so change one setting and reboot the router, then test. Here are my recommended items to test one at a time:

Wireless Network Mode > use "AC+N" instead of "AC only".

Channel Width > use "fixed 40" or "fixed 80" instead of "Dynamic..."

Extension Channel > try the other options

Airtime Fairness > try turning it off.

I do not see it on the router page you posted but you probably want to try security with WPA2 only if you have Mixed or WPA3 set.

Tell me what 5 GHZ channels are available in your country. That would help me help you. It would also be nice to know the approximate congestion levels on the available channels.

While I am at it, I'll throw some other options out:

USB WiFi adapter support on Linux has improved GREATLY over the last 5-6 years and we have many good adapters that are well supported available. Unfortuately, this chipset is not one of them so if selling your adapter to a Windows user and getting another adapter is an option, it will likely lead to far more happiness as time passes.

The site Main Menu here is:

 https://github.com/morrownr/USB-WiFi

I would recommend reading menu items 1, 2 and 4 to get an idea of how to best determine what is likely the best adapter for you. While at the Main Menu, you can use Issues to post questions. It is also a good read to see what others are talking about.

Let me know how my suggestions turn out or if you need additional info.

@morrownr

ohno1052, 2023-08-18:
@morrownr Well, any settings changes you proposed dont make network visible/connectable. (security was already wpa2 with sha256 and aes) My county is russia and avaliable channels are 32-68, 132-169.
Sreenshot from wifi analyzer below. Also, if it could help, my adapter is tplink archer t9uh. Thanks for help.

morrownr, 2023-08-18:
You might try different channels.

Maybe someone else will see this issue and have a good idea.

ohno1052, 2023-08-18:
There's something even more starnge, when I tried linux half a year ago everything was fine. Same router, same settings. Will try to play with channels now.

morrownr, 2023-08-18:
> when I tried linux half a year ago everything was fine

It is possible that it is something in the distro. My opinion is that the most solid and well supported distros are Debian 12 and Ubuntu.