morrownr/8814au #114: Error building driver when updating linux kernel

mystyq, 2023-09-15:
Hi there,

When updating the linux kernel on a debian system from `6.1.0-11` to `6.1.0-12` I get a build error.

From the make log:
```
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-12-common/scripts/Makefile.build:255: /var/lib/dkms/8814au/5.8.5.1_35583.20191029/build/os_dep/linux/rtw_proc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.1.0-12-common/Makefile:2037: /var/lib/dkms/8814au/5.8.5.1_35583.20191029/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-12-amd64'
make: *** [Makefile:2353: modules] Error 2
```

------

```
Building module:
Cleaning build area...
./dkms-make.sh.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-12-amd64 (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1_35583.20191029/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.1.0-12-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.1.0-12-amd64 (--configure):
 installed linux-image-6.1.0-12-amd64 package post-installation script subprocess returned error exit status 1
Setting up linux-headers-6.1.0-12-amd64 (6.1.52-1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.0-12-amd64.
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/8814au/5.8.5.1/source/dkms.conf)
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
```

morrownr, 2023-09-15:
Hi @mystyq 

I just tested on my main dev running kernel 6.5. Clean compile and installation so the hunt is on.

Can you post the result of:

$ dkms status

Can you run and post the result of the following commands from the driver directory:

$ make clean
$ make
If no error...
$ sudo make install
$ make clean

mystyq, 2023-09-15:
@morrownr 

`$ dkms status`
```
8812au/5.6.4.2_35491.20191025, 6.1.0-10-amd64, x86_64: installed
8812au/5.6.4.2_35491.20191025, 6.1.0-12-amd64, x86_64: installed
8814au/5.8.5.1: added
8814au/5.8.5.1_35583.20191029: added
nvidia-current/525.125.06, 6.1.0-11-amd64, x86_64: installed
nvidia-current/525.125.06, 6.1.0-12-amd64, x86_64: installed
rtl8814au/5.8.5.1, 6.1.0-11-amd64, x86_64: installed
rtl8814au/5.8.5.1, 6.1.0-12-amd64, x86_64: installed
```

`$ make clean`
`$ make`
The modules compile against `6.1.0-11`

No error - running next commands.

`$ sudo make install`
`$ make clean`

mystyq, 2023-09-15:
Here is the output when dpkg tries to install `6.1.0-12`
```
Setting up linux-image-6.1.0-12-amd64 (6.1.52-1) ...
/etc/kernel/postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.0-12-amd64.
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/8814au/5.8.5.1/source/dkms.conf)
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh.........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-12-amd64 (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1_35583.20191029/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.1.0-12-amd64 failed!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.1.0-12-amd64 (--configure):
 installed linux-image-6.1.0-12-amd64 package post-installation script subprocess returned error exit status 1
Setting up linux-headers-6.1.0-12-amd64 (6.1.52-1) ...
/etc/kernel/header_postinst.d/dkms:
dkms: running auto installation service for kernel 6.1.0-12-amd64.
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/8814au/5.8.5.1/source/dkms.conf)
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh.........(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.0-12-amd64 (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1_35583.20191029/build/make.log for more information.
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
dkms: autoinstall for kernel: 6.1.0-12-amd64 failed!
run-parts: /etc/kernel/header_postinst.d/dkms exited with return code 11
Failed to process /etc/kernel/header_postinst.d at /var/lib/dpkg/info/linux-headers-6.1.0-12-amd64.postinst line 11.
dpkg: error processing package linux-headers-6.1.0-12-amd64 (--configure):
 installed linux-headers-6.1.0-12-amd64 package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of linux-headers-amd64:
 linux-headers-amd64 depends on linux-headers-6.1.0-12-amd64 (= 6.1.52-1); however:
  Package linux-headers-6.1.0-12-amd64 is not configured yet.

dpkg: error processing package linux-headers-amd64 (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of linux-image-amd64:
 linux-image-amd64 depends on linux-image-6.1.0-12-amd64 (= 6.1.52-1); however:
  Package linux-image-6.1.0-12-amd64 is not configured yet.

dpkg: error processing package linux-image-amd64 (--configure):
 dependency problems - leaving unconfigured
Processing triggers for mailcap (3.70+nmu1) ...
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for gnome-menus (3.36.0-1.1) ...
Processing triggers for shared-mime-info (2.2-1) ...
Errors were encountered while processing:
 linux-image-6.1.0-12-amd64
 linux-headers-6.1.0-12-amd64
 linux-headers-amd64
 linux-image-amd64
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

morrownr, 2023-09-15:
You need to remove the following:

8814au/5.8.5.1: added
8814au/5.8.5.1_35583.20191029: added

Do you know how?

mystyq, 2023-09-15:
I removed both entries then upgraded the system successfully.  Thank you!  

Will this need to be done each time the kernel updates?

morrownr, 2023-09-15:
You should run the following every two months or so:

$ git pull
$ sudo sh install-driver.sh

nu11secur1ty, 2024-02-17:
> Hi there,
>
> When updating the linux kernel on a debian system from `6.1.0-11` to `6.1.0-12` I get a build error.
>
> From the make log:
>
> ```
> cc1: some warnings being treated as errors
> make[2]: *** [...]

Hi, purging the old installation and using auto detect installer official!:
OS's: Kali, Debian, Ubuntu!

### Online install:
```
curl -s https://raw.githubusercontent.com/nu11secur1ty/Kali-Linux/master/ALFA-W1F1/inst.sh | bash
```
### Online purger
```
curl -s https://raw.githubusercontent.com/nu11secur1ty/Kali-Linux/master/ALFA-W1F1/purge.sh | bash
```
BR

RacerX22288, 2024-11-04:
> You need to remove the following:
> 
> 8814au/5.8.5.1: added 8814au/5.8.5.1_35583.20191029: added
> 
> Do you know how?

I am having the same issue and I do *not* know how to remove. Halp?

morrownr, 2024-11-05:
$ sudo dkms remove -m 8814au -v 5.8.5.1_35583.20191029 --all

That was not installed with this driver.