morrownr/8814au #115: Wifi slowness with clean install running kernel 6.1.50

bdgra, 2023-11-02:
Hi, 

Hope someone can help me. 

I recently did a fresh install of Armbian 23.8.1 with kernel version 6.1.50. I did this on a new emmc module. Before I was running on Armbian 20.02.1 with kernel version 5.10.12. When I was running the old version my wifi device worked fine. I used the exact same config as I did on kernel version 5.10.12

Here is a speedtest with kernel version 6.1.50 

Testing download speed................................................................................
Download: 17.38 Mbit/s
Testing upload speed......................................................................................................
Upload: 2.23 Mbit/s

Here is a speedtest with kernel version 5.10.12

Testing download speed................................................................................
Download: 239.76 Mbit/s
Testing upload speed......................................................................................................
Upload: 219.78 Mbit/s

If I run dkms status this is the outcome 

rtl8814au/5.8.5.1, 6.1.50-current-meson64, aarch64: installed

8814au.conf (the exact same config that I was running at kernel version 5.10.12).

options 8814au rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_switch_usb_mode=1 rtw_wireless_mode=84

create_ap.conf 

CHANNEL=default
GATEWAY=10.0.10.1
WPA_VERSION=2
ETC_HOSTS=0
DHCP_DNS=gateway
NO_DNS=0
NO_DNSMASQ=0
HIDDEN=0
MAC_FILTER=0
MAC_FILTER_ACCEPT=/etc/hostapd/hostapd.accept
ISOLATE_CLIENTS=0
SHARE_METHOD=nat
IEEE80211N=1
IEEE80211AC=1
HT_CAPAB=[HT40+]
VHT_CAPAB=
DRIVER=nl80211
NO_VIRT=0
COUNTRY=
FREQ_BAND=5
NEW_MACADDR=
DAEMONIZE=0
NO_HAVEGED=0

nu11secur1ty, 2024-02-17:
You must be managing correctly your NETWORK INTERFACES! This is not a problem.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was mitigated with a runtime diagnostics+validation workflow that captures NM/RF/USB/runtime state and evaluates issue-specific pass criteria.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime healthcheck: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Runtime issue suite (#120/#117/#115/#106): https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-issue-suite.sh#L3\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.