morrownr/8814au #116: Build error on 6.7.0-0.rc0.20231104git90b0c2b2edd1.7

codebam, 2023-11-05:
```
/home/codebam/8814au make -j16
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.7.0-0.rc0.20231104git90b0c2b2edd1.7.fc40.x86_64/build M=/var/home/codebam/8814au  modules
make[1]: Entering directory '/usr/src/kernels/6.7.0-0.rc0.20231104git90b0c2b2edd1.7.fc40.x86_64'
  CC [M]  /var/home/codebam/8814au/core/rtw_cmd.o
  CC [M]  /var/home/codebam/8814au/core/rtw_security.o
  CC [M]  /var/home/codebam/8814au/core/rtw_debug.o
  CC [M]  /var/home/codebam/8814au/core/rtw_io.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ioctl_query.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ioctl_set.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ieee80211.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mlme.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mlme_ext.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mi.o
  CC [M]  /var/home/codebam/8814au/core/rtw_wlan_util.o
  CC [M]  /var/home/codebam/8814au/core/rtw_vht.o
  CC [M]  /var/home/codebam/8814au/core/rtw_pwrctrl.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rf.o
  CC [M]  /var/home/codebam/8814au/core/rtw_chplan.o
  CC [M]  /var/home/codebam/8814au/core/rtw_recv.o
  CC [M]  /var/home/codebam/8814au/core/rtw_sta_mgt.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ap.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh_pathtbl.o
In file included from ./include/linux/kernel.h:31,
                 from ./arch/x86/include/asm/percpu.h:27,
                 from ./arch/x86/include/asm/nospec-branch.h:14,
                 from ./arch/x86/include/asm/paravirt_types.h:20,
                 from ./arch/x86/include/asm/ptrace.h:97,
                 from ./arch/x86/include/asm/math_emu.h:5,
                 from ./arch/x86/include/asm/processor.h:13,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/home/codebam/8814au/include/basic_types.h:76,
                 from /var/home/codebam/8814au/include/drv_types.h:26,
                 from /var/home/codebam/8814au/core/rtw_rf.c:17:
/var/home/codebam/8814au/core/rtw_rf.c: In function ‘dump_txpwr_lmt’:
/var/home/codebam/8814au/core/rtw_rf.c:1082:67: warning: ‘strcmp’ reading 1 or more bytes from a region of size 0 [-Wstringop-overread]
 1082 |                                                                 , strcmp(ent->regd_name, rfctl->regd_name) == 0 ? rf_path_char(path) : ' ');
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/printk.h:427:33: note: in definition of macro ‘printk_index_wrap’
  427 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
      |                                 ^~~~~~~~~~~
/var/home/codebam/8814au/include/rtw_debug.h:85:26: note: in expansion of macro ‘printk’
   85 |         #define _dbgdump printk
      |                          ^~~~~~
/var/home/codebam/8814au/include/rtw_debug.h:233:25: note: in expansion of macro ‘_RTW_PRINT’
  233 |                         _RTW_PRINT(fmt, ##arg); \
      |                         ^~~~~~~~~~
/var/home/codebam/8814au/core/rtw_rf.c:1081:57: note: in expansion of macro ‘_RTW_PRINT_SEL’
 1081 |                                                         _RTW_PRINT_SEL(sel, "%3c "
      |                                                         ^~~~~~~~~~~~~~
In file included from /var/home/codebam/8814au/core/rtw_rf.c:18:
/var/home/codebam/8814au/include/hal_data.h:341:14: note: source object ‘regd_name’ of size 0
  341 |         char regd_name[0];
      |              ^~~~~~~~~
In file included from /var/home/codebam/8814au/include/drv_types.h:60:
/var/home/codebam/8814au/core/rtw_rf.c:1082:67: warning: ‘strcmp’ reading 1 or more bytes from a region of size 0 [-Wstringop-overread]
 1082 |                                                                 , strcmp(ent->regd_name, rfctl->regd_name) == 0 ? rf_path_char(path) : ' ');
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/home/codebam/8814au/include/rtw_debug.h:235:46: note: in definition of macro ‘_RTW_PRINT_SEL’
  235 |                         _seqdump(sel, fmt, ##arg) /*rtw_warn_on(1)*/; \
      |                                              ^~~
/var/home/codebam/8814au/include/hal_data.h:341:14: note: source object ‘regd_name’ of size 0
  341 |         char regd_name[0];
      |              ^~~~~~~~~
[... 62 lines cut ...]
  CC [M]  /var/home/codebam/8814au/hal/hal_btcoex_wifionly.o
  CC [M]  /var/home/codebam/8814au/hal/hal_btcoex.o
  CC [M]  /var/home/codebam/8814au/hal/hal_mp.o
  CC [M]  /var/home/codebam/8814au/hal/hal_mcc.o
  CC [M]  /var/home/codebam/8814au/hal/hal_hci/hal_usb.o
  CC [M]  /var/home/codebam/8814au/hal/led/hal_led.o
  CC [M]  /var/home/codebam/8814au/hal/led/hal_usb_led.o
  CC [M]  /var/home/codebam/8814au/hal/HalPwrSeqCmd.o
  CC [M]  /var/home/codebam/8814au/hal/rtl8814a/Hal8814PwrSeq.o
  CC [M]  /var/home/codebam/8814au/hal/rtl8814a/rtl8814a_xmit.o
  CC [M]  /var/home/codebam/8814au/hal/rtl8814a/rtl8814a_sreset.o
  CC [M]  /var/home/codebam/8814au/hal/rtl8814a/rtl8814a_hal_init.o
  CC [M]  /var/home/codebam/8814au/hal/rtl8814a/rtl8814a_phycfg.o
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/var/home/codebam/8814au/hal/led/hal_led.c: In function ‘rtw_led_set_strategy’:
/var/home/codebam/8814au/hal/led/hal_led.c:66:38: warning: implicit conversion from ‘enum _LED_STATE’ to ‘LED_CTL_MODE’ {aka ‘enum _LED_CTL_MODE’} [-Wenum-conversion]
   66 |         rtw_led_control(pri_adapter, RTW_LED_OFF);
      |                                      ^~~~~~~~~~~
make[2]: *** [/usr/src/kernels/6.7.0-0.rc0.20231104git90b0c2b2edd1.7.fc40.x86_64/Makefile:1931: /var/home/codebam/8814au] Error 2
make[1]: *** [Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.7.0-0.rc0.20231104git90b0c2b2edd1.7.fc40.x86_64'
make: *** [Makefile:2398: modules] Error 2
```

morrownr, 2023-11-05:
Hi @codebam 

I had assumed we would see the need for patches with 6.7 based on the changes that are going in. I'll start looking at this when rc1 is released next weekend.

If you run across any good patches, let me know amd thanks for the heads up.

codebam, 2023-12-11:
Hey I pulled the latest commits and tried to build. Getting this on rc4.

```
/home/codebam/8814au make -j16
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.7.0-0.rc4.20231208git5e3f5b81de80.38.fc40.x86_64/build M=/var/home/codebam/8814au  modules
make[1]: Entering directory '/usr/src/kernels/6.7.0-0.rc4.20231208git5e3f5b81de80.38.fc40.x86_64'
  CC [M]  /var/home/codebam/8814au/core/rtw_cmd.o
  CC [M]  /var/home/codebam/8814au/core/rtw_security.o
  CC [M]  /var/home/codebam/8814au/core/rtw_debug.o
  CC [M]  /var/home/codebam/8814au/core/rtw_io.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ioctl_query.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ioctl_set.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ieee80211.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mlme.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mlme_ext.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mi.o
  CC [M]  /var/home/codebam/8814au/core/rtw_wlan_util.o
  CC [M]  /var/home/codebam/8814au/core/rtw_vht.o
  CC [M]  /var/home/codebam/8814au/core/rtw_pwrctrl.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rf.o
  CC [M]  /var/home/codebam/8814au/core/rtw_chplan.o
  CC [M]  /var/home/codebam/8814au/core/rtw_recv.o
  CC [M]  /var/home/codebam/8814au/core/rtw_sta_mgt.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ap.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/home/codebam/8814au/core/rtw_xmit.o
  CC [M]  /var/home/codebam/8814au/core/rtw_p2p.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rson.o
  CC [M]  /var/home/codebam/8814au/core/rtw_tdls.o
  CC [M]  /var/home/codebam/8814au/core/rtw_br_ext.o
  CC [M]  /var/home/codebam/8814au/core/rtw_iol.o
  CC [M]  /var/home/codebam/8814au/core/rtw_sreset.o
  CC [M]  /var/home/codebam/8814au/core/rtw_btcoex_wifionly.o
  CC [M]  /var/home/codebam/8814au/core/rtw_btcoex.o
  CC [M]  /var/home/codebam/8814au/core/rtw_beamforming.o
  CC [M]  /var/home/codebam/8814au/core/rtw_odm.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rm.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rm_fsm.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rm_util.o
  CC [M]  /var/home/codebam/8814au/core/efuse/rtw_efuse.o
  CC [M]  /var/home/codebam/8814au/os_dep/osdep_service.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/os_intfs.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/usb_intf.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/ioctl_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/xmit_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/mlme_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/recv_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/wifi_regd.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_android.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_proc.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/home/codebam/8814au/hal/hal_intf.o
  CC [M]  /var/home/codebam/8814au/hal/hal_com.o
  CC [M]  /var/home/codebam/8814au/hal/hal_com_phycfg.o
  CC [M]  /var/home/codebam/8814au/hal/hal_phy.o
  CC [M]  /var/home/codebam/8814au/hal/hal_dm.o
  CC [M]  /var/home/codebam/8814au/hal/hal_dm_acs.o
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_change_beacon’:
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5155:24: error: ‘struct cfg80211_ap_update’ has no member named ‘proberesp_ies’
 5155 |                 if(info->proberesp_ies && info->proberesp_ies_len > 0) {
      |                        ^~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5155:47: error: ‘struct cfg80211_ap_update’ has no member named ‘proberesp_ies_len’
 5155 |                 if(info->proberesp_ies && info->proberesp_ies_len > 0) {
      |                                               ^~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5156:74: error: ‘struct cfg80211_ap_update’ has no member named ‘proberesp_ies’
 5156 |                         rtw_cfg80211_set_mgnt_wpsp2pie(ndev, (char *)info->proberesp_ies,
      |                                                                          ^~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5157:37: error: ‘struct cfg80211_ap_update’ has no member named ‘proberesp_ies_len’
 5157 |                                 info->proberesp_ies_len, 0x2/*PROBE_RESP*/);
      |                                     ^~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5159:24: error: ‘struct cfg80211_ap_update’ has no member named ‘assocresp_ies’
 5159 |                 if(info->assocresp_ies && info->assocresp_ies_len > 0) {
      |                        ^~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5159:47: error: ‘struct cfg80211_ap_update’ has no member named ‘assocresp_ies_len’
 5159 |                 if(info->assocresp_ies && info->assocresp_ies_len > 0) {
      |                                               ^~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:5160:74: error: ‘struct cfg80211_ap_update’ has no member named ‘assocresp_ies’
 5160 |                         rtw_cfg80211_set_mgnt_wpsp2pie(ndev, (char *)info->assocresp_ies,
      |                                                                          ^~
[... cut ...]
      |                                     ^~
make[3]: *** [scripts/Makefile.build:243: /var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/kernels/6.7.0-0.rc4.20231208git5e3f5b81de80.38.fc40.x86_64/Makefile:1929: /var/home/codebam/8814au] Error 2
make[1]: *** [Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.7.0-0.rc4.20231208git5e3f5b81de80.38.fc40.x86_64'
make: *** [Makefile:2404: modules] Error 2
```

morrownr, 2023-12-11:
Good day @codebam 

When I looked at my notes, for this adapter it said "not tested" so I accidentally uploaded what I thought was the fix without testing it. My bad. Sometimes, maybe once or twice per year, I have to rework my lab and I am in the process of doing that right now but some high priority issues came up and here I am doing the best I can. When it rains, it pours!

I took the time this morning to go over thingsfor this driver in detail and I think I have fixed the patch. It is merged. I did a compile test but have not done an operational test. Please do an operational test and report.

Remember that you can use $ git pull to update your code before compiling.

@morrownr

codebam, 2023-12-12:
Hi @morrownr, thanks so much for maintaining this driver. It compiles on the latest patch and all is working! :)

morrownr, 2023-12-12:
Thanks for letting me know. Patches for this driver always seem to be a pain in the backside. Keep an eye on it and let know if you see any problems.

nikelborm, 2024-01-25:
Hi, if this issue is still open I suppose this may be relevant to @morrownr https://github.com/tomaspinho/rtl8821ce/issues/344#issuecomment-1908735580

morrownr, 2024-01-25:
@codebam 

> Thank you very much for maintaining the driver and especially for fixing new bugs such as this one.

Thanks you for the kind words. This should be fixed. It does need to be closed as there is no reason to keep it up.

@morrownr

nu11secur1ty, 2024-02-17:
Your GCC is not compatible! You can use autodetect dependency installer ;)

### Online install:
```
curl -s https://raw.githubusercontent.com/nu11secur1ty/Kali-Linux/master/ALFA-W1F1/inst.sh | bash
```
### Online purger
```
curl -s https://raw.githubusercontent.com/nu11secur1ty/Kali-Linux/master/ALFA-W1F1/purge.sh | bash
```
BR

closed 2024-04-26 by codebam