morrownr/8814au #117: 6.5.11 Builds: Yes - Works: No

froginperil, 2023-11-19:
Fedora 39.
I am not getting any build error messages.

options 8814au rtw_drv_log_level=1 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_switch_usb_mode=0

The thing is dead. No LED.

Any suggestions?

morrownr, 2023-12-05:
Hi @froginperil 

I've been sick and unable to help.

> I am not getting any build error messages.

There may not be an errors. Did you check you vid/pid to make sure this driver supports your adapter?

Are you sure you adapter uses a rtl8814au chipset?

If secure boot turned on?

nu11secur1ty, 2024-02-17:
If someone wants to test it, just go on your OS and install it, or try to install it and report :) BR This is not a problem!

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: added runtime issue suite + expanded healthcheck and validated attached-adapter runtime visibility/RF state on rg1. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was mitigated with a runtime diagnostics+validation workflow that captures NM/RF/USB/runtime state and evaluates issue-specific pass criteria.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime healthcheck: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Runtime issue suite (#120/#117/#115/#106): https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-issue-suite.sh#L3\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.