morrownr/8814au #118: Make Build issue, CentOS 9

JolyGoodDay, 2023-11-20:
`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-325.el9.x86_64/build M=/home/pmillman/main/8814au-main  modules
make[1]: Entering directory '/usr/src/kernels/5.14.0-325.el9.x86_64'
  CC [M]  /home/pmillman/main/8814au-main/os_dep/osdep_service.o
/home/pmillman/main/8814au-main/os_dep/osdep_service.c: In function ‘thread_exit’:
/home/pmillman/main/8814au-main/os_dep/osdep_service.c:1297:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
1297 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [scripts/Makefile.build:297: /home/pmillman/main/8814au-main/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1924: /home/pmillman/main/8814au-main] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.0-325.el9.x86_64'
make: *** [Makefile:2398: modules] Error 2 `

JolyGoodDay, 2023-11-20:
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 16/16 (in-use/total processing units)
: 65411720 (total system memory)
: 5.14.0-325.el9.x86_64 (kernel version)
: gcc (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2)
: dkms-3.0.11
: ---------------------------

Checking for previously installed drivers.
rtl8814au/5.8.5.1: added
: ---------------------------

Removing a dkms installation.
Deleting module rtl8814au-5.8.5.1 completely from the DKMS tree.
Removing 8814au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8814au-5.8.5.1
Removal complete.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8814au-5.8.5.1
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
The driver was added to dkms successfully.
: ---------------------------
Sign command: /lib/modules/5.14.0-325.el9.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-325.el9.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 44.34 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.

morrownr, 2023-11-20:
Hi @JolyGoodDay 

You are not going to like what I have to say but maybe it can help you make decisions going forward.

Redhat does things to the maiinline Linux kernel that makes it non-standard. They pull changes from future kernels into the kernel they are using selectively. Supporting this would make for a LOT of work on my part and I do no even use any distros based on Redhat. I leave it open to Redhat devs to add support but the 88x2bu is the only Realtek driver here that someone has taken me up on the offer.

The easier solution could be that you go with in-kernel drivers for usb wifi solutions. I do not know what in-kernel drivers are available in your kernel but checking to see might show mt7612u or mt7610u drivers as available. Do you know how to check?

@morrownr

JolyGoodDay, 2023-11-20:
Alright no worries.

Thanks for responding so quickly!

closed 2023-11-20 by JolyGoodDay