morrownr/8814au #120: Upgrading to Kernel version 6.6.5 breaks the functionality of the driver + the entire NetworkManager

AZMindroma, 2023-12-10:
Distrubution: Arch Linux
Kernel Version: 6.6.5 (linux-zen and linux)
Desktop Environment: KDE Plasma
Tiling Window Manager: i3

Hello! So far, my experience with this driver had been extremely amazing. It even beat the actual drivers on Windows by a ton in terms of performance. 
However, when updating my system to the latest kernel version (6.6.5) today, the driver does not work in this kernel version. In fact, it breaks it quite heavily. With zen, it first it booted into KDE, then I was met with a black screen. After a restart it didn't even go into KDE, so I tried i3. In i3, I noticed no programs allowed themselves to be opened whatsoever. Even running anything as "sudo" didn't work and just resulted in an indefinite freeze of the terminal. 

Booting into linux and not zen has allowed me to access KDE, but it's not usable either. Programs don't run, sudo freezes, going into connection settings in KDE makes the KDE settings freeze, clicking the wifi symbol in the system tray freezes the taskbar, shutting down normally does not work, it's required to forcefully turn it off.

There I noticed there isn't any internet connection whatsoever. Running "ping google.com" errors with "Temporary failure in name resolution". 

When trying to access nmtui (NetworkManager), it says that the NetworkManager is not running. Stopped systemctl NetworkManager.service, stopped it, disabled it, enabled it, started it. Checked its status, online. Ran "nmtui" again, it still says that NetworkManager isn't running. Therefore I tried a fix involving /etc/resolv.conf (https://stackoverflow.com/questions/53687051/linux-command-line-error-message-temporary-failure-in-name-resolution) with no success.

When disconnecting my Netgear A7000 and rebooting, all the issues magically fixed themselves. Booting worked as usual, programs actually ran and didn't freeze my system, sudo worked, shutting down the system worked without it being done forcefully. To see if it isn't a faulty NetworkManager, I've established an USB-tethering connection, which worked perfectly, didn't stop programs from not running and let me open "nmtui" or the connection settings in KDE. Upon plugging in the netgear, everything breaks again.

Noticing it has something to do with the wifi, I reinstalled the driver in hopes of fixing the problem, but these problems persisted. Now, I'm not intelligent enough to both analyze the new kernel and the driver to notice any conflicts, so I'd like to ask if anyone here or/and morrownr have any knowledge on how to possibly fix this issue (switching to another distribution or switching the netgear would be the last option). 

I've tried making my issue as understandable and detailed as possible, but if more detail is needed, please let me know. Thanks a lot in advance for any help!

morrownr, 2023-12-10:
Try:

$ git pull
$ sudo sh install-driver.sh

Flopblopper, 2023-12-11:
I can confirm that I'm having the same issues on Debian 12.4. Installing the latest version of this driver causes all sorts of strange behavior.

AZMindroma, 2023-12-11:
> Try:
> 
> $ git pull $ sudo sh install-driver.sh

Thank you for your reply! I've done this, however, the issue seems to persist. 

> I can confirm that I'm having the same issues on Debian 12.4. Installing the latest version of this driver causes all sorts of strange behavior.

Your kernel version is 6.1.0 then, isn't it? I hadn't updated the driver for a long time since it always worked, and even without the newest driver version, upgrading to kernel 6.6.5 had caused these issues (and they persisted even after upgrading/reinstalling). This raises the question what exactly is the troublemaker here.

Flopblopper, 2023-12-11:
That is correct. I believe the problem is with the driver, rather than any particular kernel.
I had gotten into the habit of updating this driver any time Debian gets a kernel update, and after the most recent, I began experiencing the exact same issues you've described.

Flopblopper, 2023-12-11:
Update:
I just came across a post on Reddit complaining about stability and Wi-Fi issues with the latest Debian kernel, so perhaps my own issues didn't have anything to do with this driver after all. Apologies for the confusion.

AZMindroma, 2023-12-11:
I've also found a forum entry now: https://bbs.archlinux.org/viewtopic.php?id=290976&p=2

> I can also report this is an issue with devices using the following chipset:
> rtl8814au
> rtl88x2bu

The issue does indeed seem to relate to the chipset we are using, but it seems like it's an issue with the kernel and not the driver. I also apologize for any confusion. The following kernel (6.6.6) includes a patch for this error, so I'll close this issue. https://github.com/archlinux/linux/commit/9099d0682462e8664c8cbbe8989de65e50f62b17

morrownr, 2023-12-11:
@AZMindroma @Flopblopper 

I reopened this issue as others will likely have similar problems over the next few days.

I just spent the morning fixed and merging fixes for kernel 6.7 support. It seems that when it rains, it pours. I'm seeing several reports on several of the drivers I have up here. My best guess based on the reports is that there is a backport that has gone into multiple kernels that has broken these drivers badly. I'm seeing kernel 6.1.65 working fine and 6.1.66 broken in my own testing. Ubuntu users reporting the last kernel 6.2.x is broken probably because the Ubuntu devs applied the problem backport.

The link for Arch that you posted very well could be the problem and it is nasty. That is not something I can work around in these drivers so, until this is fixed upstream and has time to flow out to us, your best bet is to backup to a kernel that works and stay there until such time as the fix is in. Thanks for the info.

@morrownr

sqrt-7, 2023-12-12:
Had the same issue, updating my Arch distro to kernel 6.6.5 made network manager hard lock my system. I think the bug was backported to the LTS kernel because that also broke.

I've just updated to kernel 6.6.6 + the latest commit of this repo, and the issue seems to be gone.

morrownr, 2023-12-12:
Thanks for the report. 

> I think the bug was backported to the LTS kernel because that also broke.

Indeed. Why break one kernel when you can break them all?

mystyq, 2023-12-15:
It appears the issue is now fixed, Debian 12 `6.1.0-16-amd64`

nu11secur1ty, 2024-02-17:
This is not an issue!