morrownr/8814au #122: Error during compilation under Truenas Scale 6.1.63-production+truenas

m1sl, 2024-01-27:
I have a problem compiling the driver under the Truenas Scale (kernel 6.1.63-production+truenas). The error message is the following:

`/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c: In function 'rtw_ndev_destructor':
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:4208:19: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
 4208 |         if (ndev->ieee80211_ptr)
      |                   ^~~~~~~~~~~~~
      |                   ieee802154_ptr
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:17:
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:4209:39: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
 4209 |                 rtw_mfree((u8 *)ndev->ieee80211_ptr, sizeof(struct wireless_dev));
      |                                       ^~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/osdep_service.h:276:53: note: in definition of macro 'rtw_mfree'
  276 | #define rtw_mfree(pbuf, sz)             _rtw_mfree((pbuf), (sz))
      |                                                     ^~~~
At top level:
cc1: note: unrecognized command-line option '-Wno-enum-int-mismatch' may have been intended to silence earlier diagnostics
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2015: /var/lib/dkms/rtl8814au/5.8.5.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.63-production+truenas'
make: *** [Makefile:2404: modules] Error 2`

The error is not sensitive to the availability of the dkms framework.
Any suggestion is highly appreciated.

morrownr, 2024-01-28:
Hi @m1sl 

I found the TrueNAS website. It says TrueNAS is based on Debian but does not say which version. I'm in the blind here.

The `install-driver.sh` script is designed to put a log on the screen as it runs. If you could run it and post what appears on the screen, that might help.

@morrownr

m1sl, 2024-01-28:
Hi @morrownr 

I posted the part of this log where the script encountered an error. I apologize for its formatting, I tried to make it more readable, but unfortunately here in GitHub it looks like a total mess. Anyway here is the full log of the script itself which was displayed on the screen during its run:

```
root@test[~/8814au]# ./install-driver.sh
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 4/4 (in-use/total processing units)
: 4008912 (total system memory)
: 6.1.63-production+truenas (kernel version)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
EFI variables are not supported on this system
EFI variables are not supported on this system
EFI variables are not supported on this system
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8814au-5.8.5.1
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Building module:
Cleaning build area...
./dkms-make.sh.......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.63-production+truenas (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```

I think I found the way to post it correctly) So please find it below in a more readable form:

```
root@test[~/8814au]# cat  /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log
DKMS make.log for rtl8814au-5.8.5.1 for kernel 6.1.63-production+truenas (x86_64)
Sat Jan 27 15:55:13 PST 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.63-production+truenas/build M=/var/lib/dkms/rtl8814au/5.8.5.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.63-production+truenas'
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_recv.o
[... 20 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_ops_linux.o
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c: In function 'rtw_ndev_destructor':
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:4208:19: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
 4208 |         if (ndev->ieee80211_ptr)
      |                   ^~~~~~~~~~~~~
      |                   ieee802154_ptr
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:17:
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:4209:39: error: 'struct net_device' has no member named 'ieee80211_ptr'; did you mean 'ieee802154_ptr'?
 4209 |                 rtw_mfree((u8 *)ndev->ieee80211_ptr, sizeof(struct wireless_dev));
      |                                       ^~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/osdep_service.h:276:53: note: in definition of macro 'rtw_mfree'
  276 | #define rtw_mfree(pbuf, sz)             _rtw_mfree((pbuf), (sz))
      |                                                     ^~~~
At top level:
cc1: note: unrecognized command-line option '-Wno-enum-int-mismatch' may have been intended to silence earlier diagnostics
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:2015: /var/lib/dkms/rtl8814au/5.8.5.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.63-production+truenas'
make: *** [Makefile:2404: modules] Error 2
```

morrownr, 2024-01-29:
@m1sl 

I just compiled this driver on my main dev box which runs Debian 12:

```
$ sudo sh install-driver.sh
[sudo] password for morrow:         
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 4/4 (in-use/total processing units)
: 12208056 (total system memory)
: 6.1.0-17-amd64 (kernel version)
: gcc (Debian 12.2.0-14) 12.2.0
: dkms-3.0.10
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8814au-5.8.5.1
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/lib/linux-kbuild-6.1/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
./dkms-make.sh..............
Signing module /var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.ko
Cleaning build area...
Compile time: 122.54 seconds
The driver was built by dkms successfully.
: ---------------------------

8814au.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.0-17-amd64/updates/dkms/
depmod....
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

```

Installation here was smooth. My best guess is that the problem is something the TrueNAS devs have done. I don't have time to install and learn TrueNAS right now so you might want to take this issue to the TrueNAS forums to see if anyone has any ideas over there.

@morrownr

nu11secur1ty, 2024-02-17:
> I have a problem compiling the driver under the Truenas Scale (kernel 6.1.63-production+truenas). The error message is the following:
>
> `/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c: [...]

Please, check your package and software dependencies! BR

mlsok, 2024-02-17:
> Please, check your package and software dependencies! BR

All dependencies are met! Except for the kernel headers. I found 3 different types in my system. One of them is without the suffix "production+truenas". The wifi module can be compiled against it, but obviously, it can not be used by the system which is using one of the headers marked by "production+truenas". So the problem is in the headers developed for the TrueNAS. Unfortunately, I was not able to find support on the TrueNAS forum. There was one interesting advice to use a wifi bridge, but in the end, I ran a cable to my router and unplugged the WiFi card :)

nu11secur1ty, 2024-02-21:
> All dependencies are met! Except for the kernel headers. I found 3 different types in my system. One of them is without the suffix "production+truenas". The wifi module can be compiled against it, [...]

Good work, I will see about that. BR

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: removed direct `net_device.ieee80211_ptr` destructor access; rely on cfg80211 resource unregister/free paths. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.