morrownr/8814au #124: Less catching with 8814au

VeritasLibera, 2024-01-31:
Hello:) I installed https://github.com/morrownr/8814au/ on rpi4 (Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux )

When I start airodump-ng with AWUS1900  (sudo airodump-ng wlan1) It make much less results (I mean beacons) as TL-WN722N (v1.0).
When I am starting airodump-ng repeatedly TL-WN722N has no problem with that. AWUS1900 show less and less info (low beacons or or no APs (even if there are APs).

Sometimes when I restart rpi4, AWUS1900 make normal result with standard catch of beacons (in the first scan after reboot).

On the pictures is airodump-ng on TL-WN722N and AWUS1900 (in the same time period is different beacons catch)

[image]
[image]

morrownr, 2024-01-31:
Hi @VeritasLibera 

> When I start airodump-ng with AWUS1900...

Well, if you are working wireless security/pen testing, I recommend you use a different adapter with a chipset and driver that does a good job. There are no Realtek drivers that do this. Your best bets are the mt7610u, mt7612u and mt7921au chipsets. The Alfa ACHM is very highly rated for this use. It supports monitor mode and `active` monitor mode. I can recommend others but you can go to the site Main Menu and read menu items 1 and 2 which should provide a lot of information:

https://github.com/morrownr/USB-WiFi

@morrownr

VeritasLibera, 2024-02-01:
Thanks for advice, I already ordered AWUS036ACHM. However I found a solution for AWUS1900. I have to make:
1. airmon-ng check kill 
2. unplug adapter, plug adapter (AWUS1900) (the adapter should not light up or flash)
3. airmon-ng start wlan0 
and its work totally great.
I hope this helps someone.

morrownr, 2024-02-01:
@VeritasLibera 

You might want to try the script I made to help users get into monitor mode:

https://github.com/morrownr/Monitor_Mode

The primary difference between start-mon.sh and `airmon-ng check kill` is that airmon-ng tries to kill interfering processes whereas start-mon.sh pauses the process. I found that it is very difficult to kill some processes in that they have code to detect when they are killed and they rapidly restart. However, if the same processes are paused, there is nothing to restart and they cannot cause problems because they are not active. Try it.

FYI: If you like to work on scripts, start-mon.sh might be a good project for you. There are a lot of improvements that could be made if someone has time. I think having a menu system would be cool.

@morrownr

nu11secur1ty, 2024-02-17:
> Hello:) I installed https://github.com/morrownr/8814au/ on rpi4 (Linux raspberrypi 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux )
> 
> When I start airodump-ng with AWUS1900 (sudo airodump-ng wlan1) It make much less results (I mean beacons) as TL-WN722N (v1.0). When I am starting airodump-ng repeatedly TL-WN722N has no problem with that. AWUS1900 show less and less info (low beacons or or no APs (even if there are APs).
> 
> Sometimes when I restart rpi4, AWUS1900 make normal result with standard catch of beacons (in the first scan after reboot).
> 
> On the pictures is airodump-ng on TL-WN722N and AWUS1900 (in the same time period is different beacons catch). When adapters catching much longer (both adapters working at same time) TL-WN722N catch much more beacons than AWUS1900.
> 
> AWUS1900 ![01](https://private-user-images.githubusercontent.com/93779994/301136284-3306d705-212c-4e1a-aec5-757db2fc3a21.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxNDk2NjQsIm5iZiI6MTcwODE0OTM2NCwicGF0aCI6Ii85Mzc3OTk5NC8zMDExMzYyODQtMzMwNmQ3MDUtMjEyYy00ZTFhLWFlYzUtNzU3ZGIyZmMzYTIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE3VDA1NTYwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk5ZjBjYWYxNDU5ZTI3YjkwOGEzMDQyNWY3YWM0ODI0MmExMWU1OWVhZWZjMTVjZDhjZmYxYmVhNTVkZmYwMGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pgX7wYExuIjitmBwmNSNb0C3d67eN4Pz4N7_G0n6Vmg)
> 
> TL-WN722N ![02](https://private-user-images.githubusercontent.com/93779994/301136296-5403ccaf-b5f6-4ea2-bf7d-663e42b60f37.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxNDk2NjQsIm5iZiI6MTcwODE0OTM2NCwicGF0aCI6Ii85Mzc3OTk5NC8zMDExMzYyOTYtNTQwM2NjYWYtYjVmNi00ZWEyLWJmN2QtNjYzZTQyYjYwZjM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE3VDA1NTYwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE1ZDBiY2RiZjk3ODYxNzQ1ZjQyMWNkNWQxYTU4OTMxZDY2MTFlZmUyZDg5ZGE1NjU4ZDVhZmRmZjk5NTI4YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.v6eaIr7eMZUJhknxlCq1Cw5mQ-yM4OumK1-OiS_N3cY)

And where is the problem? :D