morrownr/8814au #127: Unable to build this module on Kernel 6.1

MR-Diamond, 2024-03-07:
OS: Manjaro Linux
Module version: rtl8814au 5.8.5.1.r149.g752d8ea

As DKMS attempts to build the module on Kernel 6.1, it stops at:

```
dkms install --no-depmod rtl8814au/5.8.5.1.r149.g752d8ea -k 6.1.80-1-MANJARO
Error! Bad return status for module build on kernel: 6.1.80-1-MANJARO (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/make.log for more information.
WARNING: `dkms install --no-depmod rtl8814au/5.8.5.1.r149.g752d8ea -k 6.1.80-1-MANJARO' exited 10
```

All is fine, instead, with Kernel 5.10.211

Content of make.log while building for Kernel 6.1:

```
DKMS make.log for rtl8814au-5.8.5.1.r149.g752d8ea for kernel 6.1.80-1-MANJARO (x86_64)
gio 7 mar 2024, 14:51:19, CET
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.1.80-1-MANJARO/build M=/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build  modules
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ap.o
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_sta_mgt.c: In function ‘rtw_mfree_stainfo’:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_sta_mgt.c:384:25: warning: the comparison will always evaluate as ‘true’ for the address of ‘lock’ will never be NULL [-Waddress]
  384 |         if (&psta->lock != NULL)
      |                         ^~
In file included from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/drv_types.h:103,
                 from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_sta_mgt.c:17:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/sta_info.h:263:17: note: ‘lock’ declared here
  263 |         _lock   lock;
      |                 ^~~~
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_rson.o
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ap.c: In function ‘rtw_ap_update_chbw_by_ifbmp’:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ap.c:4403:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4403 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
In file included from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ap.c:17:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/drv_types.h:1112:19: note: ‘padapters’ declared here
 1112 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_ap.c:4423:42: warning: the comparison will always evaluate as ‘true’ for the address of ‘padapters’ will never be NULL [-Waddress]
 4423 |                 if (!(ifbmp & BIT(i)) || !dvobj->padapters)
      |                                          ^
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/drv_types.h:1112:19: note: ‘padapters’ declared here
 1112 |         _adapter *padapters[CONFIG_IFACE_NUMBER];/*IFACE_ID_MAX*/
      |                   ^~~~~~~~~
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/linux/usb_intf.o
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.c:2566:28: warning: passing argument 1 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 2566 |         _rtw_memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                     ~~~~~~~^~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.c:922:24: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  922 | void _rtw_memcpy(void *dst, const void *src, u32 sz)
      |                  ~~~~~~^~~
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.c: In function ‘rtw_random32’:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.c:2690:16: error: implicit declaration of function ‘prandom_u32’; did you mean ‘prandom_u32_max’? [-Werror=implicit-function-declaration]
 2690 |         return prandom_u32();
      |                ^~~~~~~~~~~
      |                prandom_u32_max
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/osdep_service.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/linux/os_intfs.c: In function ‘rtw_os_ndev_register’:
/var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/linux/os_intfs.c:1900:9: error: too many arguments to function ‘netif_napi_add’
 1900 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
      |         ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/osdep_service_linux.h:31,
                 from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/osdep_service.h:57,
                 from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/linux/os_intfs.c:17:
./include/linux/netdevice.h:2575:1: note: declared here
 2575 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:2010: /var/lib/dkms/rtl8814au/5.8.5.1.r149.g752d8ea/build] Error 2
make: *** [Makefile:2384: modules] Error 2

```

morrownr, 2024-03-07:
Hi @MR-Diamond 

I'll do some compile tests on kernels 6.1 and 6.6 this afternoon as I have time. Have you updated the driver recently?

$ git pull
$ sudo sh install-driver.sh

MR-Diamond, 2024-03-09:
I've built the module from the latest commit. The error didn't appeared again: now I am on Kernel 6.1 with functional module; I close this ticket.

closed 2024-03-09 by MR-Diamond

morrownr, 2024-03-09:
That problem did look like you were compiling dated code. Remember that this driver is not stagnant. It has to be updated regularly just to keep up with the Linux kernel.

FYI: In case you are interested, here is the Main Menu for this site. It may have info that is of use to you:

 https://github.com/morrownr/USB-WiFi