morrownr/8814au #128: build errors on 6.9.0 rc0

codebam, 2024-03-16:
Here are the build errors on 6.9.0 rc0. Just leaving this here in case it helps. Cheers :beer:

```
/home/codebam/8814au make -j16
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.9.0-0.rc0.20240315gite5eb28f6d1af.7.fc41.x86_64/build M=/var/home/codebam/8814au  modules
make[1]: Entering directory '/usr/src/kernels/6.9.0-0.rc0.20240315gite5eb28f6d1af.7.fc41.x86_64'
  CC [M]  /var/home/codebam/8814au/core/rtw_cmd.o
  CC [M]  /var/home/codebam/8814au/core/rtw_security.o
  CC [M]  /var/home/codebam/8814au/core/rtw_debug.o
  CC [M]  /var/home/codebam/8814au/core/rtw_io.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ioctl_query.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ioctl_set.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ieee80211.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mlme.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mlme_ext.o
  CC [M]  /var/home/codebam/8814au/core/rtw_mi.o
  CC [M]  /var/home/codebam/8814au/core/rtw_wlan_util.o
  CC [M]  /var/home/codebam/8814au/core/rtw_vht.o
  CC [M]  /var/home/codebam/8814au/core/rtw_pwrctrl.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rf.o
  CC [M]  /var/home/codebam/8814au/core/rtw_chplan.o
  CC [M]  /var/home/codebam/8814au/core/rtw_recv.o
  CC [M]  /var/home/codebam/8814au/core/rtw_sta_mgt.o
  CC [M]  /var/home/codebam/8814au/core/rtw_ap.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/home/codebam/8814au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/home/codebam/8814au/core/rtw_xmit.o
  CC [M]  /var/home/codebam/8814au/core/rtw_p2p.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rson.o
  CC [M]  /var/home/codebam/8814au/core/rtw_tdls.o
  CC [M]  /var/home/codebam/8814au/core/rtw_br_ext.o
  CC [M]  /var/home/codebam/8814au/core/rtw_iol.o
  CC [M]  /var/home/codebam/8814au/core/rtw_sreset.o
  CC [M]  /var/home/codebam/8814au/core/rtw_btcoex_wifionly.o
  CC [M]  /var/home/codebam/8814au/core/rtw_btcoex.o
  CC [M]  /var/home/codebam/8814au/core/rtw_beamforming.o
  CC [M]  /var/home/codebam/8814au/core/rtw_odm.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rm.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rm_fsm.o
  CC [M]  /var/home/codebam/8814au/core/rtw_rm_util.o
  CC [M]  /var/home/codebam/8814au/core/efuse/rtw_efuse.o
  CC [M]  /var/home/codebam/8814au/os_dep/osdep_service.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/os_intfs.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/usb_intf.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/ioctl_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/xmit_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/mlme_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/recv_linux.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/wifi_regd.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_android.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_proc.o
  CC [M]  /var/home/codebam/8814au/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/home/codebam/8814au/hal/hal_intf.o
  CC [M]  /var/home/codebam/8814au/hal/hal_com.o
  CC [M]  /var/home/codebam/8814au/hal/hal_com_phycfg.o
  CC [M]  /var/home/codebam/8814au/hal/hal_phy.o
  CC [M]  /var/home/codebam/8814au/hal/hal_dm.o
  CC [M]  /var/home/codebam/8814au/hal/hal_dm_acs.o
  CC [M]  /var/home/codebam/8814au/hal/hal_btcoex_wifionly.o
  CC [M]  /var/home/codebam/8814au/hal/hal_btcoex.o
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:437:17: error: too many arguments to function ‘cfg80211_ch_switch_started_notify’
  437 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, 0, false, 0);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/home/codebam/8814au/hal/hal_mp.o
In file included from /var/home/codebam/8814au/include/osdep_service_linux.h:94,
                 from /var/home/codebam/8814au/include/osdep_service.h:57,
                 from /var/home/codebam/8814au/include/drv_types.h:27,
                 from /var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:8809:6: note: declared here
 8809 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.c:456:9: error: too many arguments to function ‘cfg80211_ch_switch_notify’
  456 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef, 0, 0);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/net/cfg80211.h:8793:6: note: declared here
 8793 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /var/home/codebam/8814au/hal/hal_mcc.o
  CC [M]  /var/home/codebam/8814au/hal/hal_hci/hal_usb.o
make[3]: *** [scripts/Makefile.build:243: /var/home/codebam/8814au/os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/kernels/6.9.0-0.rc0.20240315gite5eb28f6d1af.7.fc41.x86_64/Makefile:1930: /var/home/codebam/8814au] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.9.0-0.rc0.20240315gite5eb28f6d1af.7.fc41.x86_64'
make: *** [Makefile:2395: modules] Error 2
```

morrownr, 2024-03-17:
Hi @codebam 

Any chance of getting you to back off to kernel 6.8? I worked on getting the driver to work on 6.8 but do not think I finished and I would like to see what you see regarding 6.8. I need to get it working on 6.8 and then we can worry about 6.8.

FYI: This specific driver is not Realtek's best effort. It is hard to maintain.

codebam, 2024-03-17:
It builds and works on 6.8.0-0.rc7.20240308git3aaa8ce7a335.60.fc41.x86_64 :+1:

morrownr, 2024-03-17:
Okay. Thanks. I'll take a look but am busy right now so this may take some time.

codebam, 2024-03-17:
Okay, no rush I have another working wireless card if I really want to use the latest kernel. I'll use 6.8.0 rc7 though, this card is better. Thanks again

morrownr, 2024-03-27:
@codebam 

I think I have the proper updates in. Please test and report.

$ git pull
$ sudo sh install-driver.sh

FYI: The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 have information about USB WiFi adapters on Linux. This might come in handy if you decide to get a faster and easier to use adapter on Linux at some point.

codebam, 2024-03-28:
It builds successfully on 6.9.0-0.rc1. Thanks!

I'll check out that list if I'm looking to buy a new wireless card. The one I have now is fine, I think most of the issue with my speeds is the distance to the access point.

closed 2024-04-26 by codebam