morrownr/8814au #129: Alpine virt 3.19.1 x86_64 build error

RGM36, 2024-03-23:
Hello,

I just catch a build error on Alpine Linux "virt" 3.19.1 (x86_64)

My build packages:
- linux-virt-dev
- alpine-sdk
- linux-lts-dev
- linux-headers

Output:
redbox:/home/red/8814au# ./install-driver.sh 
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 4/4 (in-use/total processing units)
: 8048204 (total system memory)
: 6.6.22-0-virt (kernel version)
: gcc (Alpine 13.2.1_git20231014) 13.2.1 20231014
: mokutil not installed
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The non-dkms installation routines are in use.
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.6.22-0-virt/build M=/home/red/8814au  modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.22-0-virt'
  CC [M]  /home/red/8814au/core/rtw_cmd.o
  CC [M]  /home/red/8814au/core/rtw_security.o
  CC [M]  /home/red/8814au/core/rtw_debug.o
  CC [M]  /home/red/8814au/core/rtw_io.o
  CC [M]  /home/red/8814au/core/rtw_ioctl_query.o
  CC [M]  /home/red/8814au/core/rtw_ioctl_set.o
  CC [M]  /home/red/8814au/core/rtw_ieee80211.o
  CC [M]  /home/red/8814au/core/rtw_mlme.o
  CC [M]  /home/red/8814au/core/rtw_mlme_ext.o
  CC [M]  /home/red/8814au/core/rtw_mi.o
  CC [M]  /home/red/8814au/core/rtw_wlan_util.o
  CC [M]  /home/red/8814au/core/rtw_vht.o
  CC [M]  /home/red/8814au/core/rtw_pwrctrl.o
  CC [M]  /home/red/8814au/core/rtw_rf.o
  CC [M]  /home/red/8814au/core/rtw_chplan.o
  CC [M]  /home/red/8814au/core/rtw_recv.o
  CC [M]  /home/red/8814au/core/rtw_sta_mgt.o
  CC [M]  /home/red/8814au/core/rtw_ap.o
  CC [M]  /home/red/8814au/core/mesh/rtw_mesh.o
  CC [M]  /home/red/8814au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/red/8814au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/red/8814au/core/rtw_xmit.o
  CC [M]  /home/red/8814au/core/rtw_p2p.o
  CC [M]  /home/red/8814au/core/rtw_rson.o
  CC [M]  /home/red/8814au/core/rtw_tdls.o
  CC [M]  /home/red/8814au/core/rtw_br_ext.o
  CC [M]  /home/red/8814au/core/rtw_iol.o
  CC [M]  /home/red/8814au/core/rtw_sreset.o
  CC [M]  /home/red/8814au/core/rtw_btcoex_wifionly.o
  CC [M]  /home/red/8814au/core/rtw_btcoex.o
  CC [M]  /home/red/8814au/core/rtw_beamforming.o
  CC [M]  /home/red/8814au/core/rtw_odm.o
  CC [M]  /home/red/8814au/core/rtw_rm.o
  CC [M]  /home/red/8814au/core/rtw_rm_fsm.o
  CC [M]  /home/red/8814au/core/rtw_rm_util.o
  CC [M]  /home/red/8814au/core/efuse/rtw_efuse.o
  CC [M]  /home/red/8814au/os_dep/osdep_service.o
  CC [M]  /home/red/8814au/os_dep/linux/os_intfs.o
  CC [M]  /home/red/8814au/os_dep/linux/usb_intf.o
  CC [M]  /home/red/8814au/os_dep/linux/usb_ops_linux.o
/home/red/8814au/os_dep/linux/os_intfs.c: In function 'rtw_ndev_destructor':
/home/red/8814au/os_dep/linux/os_intfs.c:4212:17: error: 'struct net_device' has no member named 'ieee80211_ptr'
 4212 |         if (ndev->ieee80211_ptr)
      |                 ^~
In file included from /home/red/8814au/include/drv_types.h:27,
                 from /home/red/8814au/os_dep/linux/os_intfs.c:17:
/home/red/8814au/os_dep/linux/os_intfs.c:4213:37: error: 'struct net_device' has no member named 'ieee80211_ptr'
 4213 |                 rtw_mfree((u8 *)ndev->ieee80211_ptr, sizeof(struct wireless_dev));
      |                                     ^~
/home/red/8814au/include/osdep_service.h:276:53: note: in definition of macro 'rtw_mfree'
  276 | #define rtw_mfree(pbuf, sz)             _rtw_mfree((pbuf), (sz))
      |                                                     ^~~~
make[3]: *** [scripts/Makefile.build:243: /home/red/8814au/os_dep/linux/os_intfs.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-headers-6.6.22-0-virt/Makefile:1913: /home/red/8814au] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.22-0-virt'
make: *** [Makefile:2395: modules] Error 2
An error occurred:  2

morrownr, 2024-03-23:
@RGM36 

Hello,

After seeing your report, I did test compiles with Debian 12 and RasPiOS, both using kernel 6.6, I did not see any problems. This is a fairly heavily used distro in that it averages around 110 downloads per day and I am not seeing reports similar to this. I think this is going to be an Alpine specific problem and Alpine is not one of the distros I have on my test systems. Recommend you ask on the Alpine forums. This may be a problem with the Alpine header files.

Regards,

@morrownr

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: removed `bc` dependency from both Makefile GCC version check and install script preflight for minimal environments. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.