morrownr/8814au #130: impossible/error driver update Fedora 6.8.4-200

dakkar96, 2024-04-07:
Hi everyone, 

Sorry, big noob here. After upgrading to Fedora version "6.8.4-200.fc39.x86_64", my USB-WiFi dongle is not recognized anymore (TP-Link T9UH) .

I tried to re-install and it seemed to work fine in the Terminal since I have this message after writing "dkms status" command :

rtl8814au/5.8.5.1, 6.8.4-200.fc39.x86_64, x86_64: installed

But I still don't have wifi because my usb dongle is not recognized, even after multiple reboots ...
And when I jump back to an old Fedora version such as 6.7.11 everything is ok, my usb is recognized

Any idea how to solve this ? 

Many thanks in advance

morrownr, 2024-04-07:
Run `sudo sh install-driver.sh` again and post the onscreen log here.

dakkar96, 2024-04-08:
[image]
[image]

I didn't make a reboot at this point for the screenshot and when I did it didn't work either.

morrownr, 2024-04-08:
Reboot and then paste the output of:

$ dkms status

dakkar96, 2024-04-09:
Ok I ran the remove-driver.sh script and then I processed a manual installation like this :
(in my driver folder)
$ make clean
$ make
$ sudo make install
$ sudo reboot

no errors during the process, but still no wifi after rebooting.

and weird thing, this is my dkms status right now :
[image]

is this normal ? what should I do ?

morrownr, 2024-04-09:
> is this normal ?

Yes, dkms status should show nothing since we ran remove-driver.sh.

> I ran the install-driver script. I still don't have wifi on this, but it came back for previous versions.

This is beginning to sound like a bug in fedora. Both methods of installation should have installed the driver to the currently active kernel. Since it is installing it to an older, inactive kernel, the question is why? You might want to present this info to a fedora forum to see what you get.

dakkar96, 2024-04-14:
Everything came back to normal with update to 6.8.5-201

morrownr, 2024-04-14:
That is good.

morrownr, 2024-04-26:
@dakkar96 

Are you still stuck? If so, I have something I will merge so you can try it.

@morrownr

dakkar96, 2024-04-29:
Hi, everything is working fine here with the driver. I am currently running Fedora 40 without issue. Thank you!

morrownr, 2024-04-29:
Thanks for letting me know. If you running into problems as kernel upgrades flow into your system, let me know.

dakkar96, 2025-04-25:
Hi there ! It seems that the driver doesn't work with Fedora 42 (6.14.3-300)

Any idea how to solve this ? 

Thanks in advance

morrownr, 2025-04-25:
Hi @dakkar96 

I am in the process of discontinuing updates on the driver in this repo as we now have merged a new driver in the mainline Linux kernel. It went into kernel 6.15. What you can do is uninstall the driver in this repo:

$ sudo sh remove-driver.sh

...and then go to the following repo which is where we work on the new driver:

https://github.com/lwfinger/rtw88

Install that driver as it will work on your distro and when you get to kernel 6.15 or later, your adapter will be plug and play. WooHoo!

If you have problems with the rtw88 driver, open an issue in that repo.

Let me know how it goes.

dakkar96, 2025-04-25:
everything works fine ! top project ! many thanks :)

morrownr, 2025-04-25:
You are welcome. The devs that work on that driver watch issues in that rtw88 repo so if you have any problems, open an issue. See you over there.

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: DKMS staging cleanup + compiler-flag compatibility probing reduce kernel-update build failures in Fedora-like environments. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.