morrownr/8814au #133: 5ghz missing/doesnt work

k1LLUM1N4T1, 2024-05-13:
I upgraded kernel version 6.5 to 6.6 . 5ghz missed i cant use anymore. Only 2.4ghz working.
awus1900

i deleted and install driver again but still doesnt work.

[image]

help me please

morrownr, 2024-05-13:
I can do some testing here to see what I can find out. What distro are your using?

k1LLUM1N4T1, 2024-05-13:
> I can do some testing here to see what I can find out. What distro are your using?

kali 2024.1

k1LLUM1N4T1, 2024-05-14:
> I can do some testing here to see what I can find out. What distro are your using?

any news?

RootLUG, 2024-05-14:
I installed the drivers from git around a week ago both on kali and arch linux and 5ghz network was working fine in monitor mode (AWUS 1900)

morrownr, 2024-05-14:
@k1LLUM1N4T1 

I installed this driver on Debian 12 yesterday and I am not seeing the problem. I am connected to and working with an AP using 5 GHz. Could you run the following?

$ iw list

... pull the 5 GHz channel section out and post it here. It should look something like the following:

```
			* 5180 MHz [36] (23.0 dBm)
			* 5200 MHz [40] (23.0 dBm)
			* 5220 MHz [44] (23.0 dBm)
			* 5240 MHz [48] (23.0 dBm)
			* 5260 MHz [52] (24.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (24.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (24.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (24.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (24.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (24.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (24.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (24.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (24.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (24.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (24.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (24.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (24.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (24.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (24.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
			* 5885 MHz [177] (disabled)

```

k1LLUM1N4T1, 2024-05-14:
> I installed the drivers from git around a week ago both on kali and arch linux and 5ghz network was working fine in monitor mode (AWUS 1900)

can you use tools normally? for example airgeddon.

> @k1LLUM1N4T1
>
> I installed this driver on Debian 12 yesterday and I am not seeing the problem. I am connected to and working with an AP using 5 GHz. Could you run the following?
>
> $ iw list
>
> ... pull [...]
[image]

MONITORING
[image]

When try to attack on 5ghz:
[image]

i installed fresh kali but its still same

morrownr, 2024-05-14:
> i installed fresh kali but its still same...

I used to keep a partition on a test system for testing Kali but they pissed me off last year. I have little tolerance for excessive levels of bugs. Test it on Debian 12 or Ubuntu or basically anything else and get back to me.

Side note based on your use case: You might save yourself a lot of headaches moving your Kali work to an adapter like an Alfa ACHM. My Main Menu here is:

https://github.com/morrownr/USB-WiFi

Scroll down to see the Main Menu. Menu item 2, Plug and Play List has a lot of adapters that should work well with Kali. One that is a favorite is the Alfa ACHM. If you have questions, ask.

FYI: the driver in this repo is not that great. Realtek simply does not care to support Linux. Mediatek is supporting far better drivers for Linux.

k1LLUM1N4T1, 2024-05-15:
> I used to keep a partition on a test system for testing Kali but they pissed me off last year. I have little tolerance for excessive levels of bugs. Test it on Debian 12 or Ubuntu or basically [...]

Working normally on ParrotOS. 
Awus1900 really sucks. I'm sorry for wasting your time on an unnecessary and annoying issue.
Finally, which is the current and most compatible adapter you can recommend for pentest? Is AWUS036AXML good choice?
Thanks for your patience🙌

morrownr, 2024-05-15:
>  I'm sorry for wasting your time on an unnecessary and annoying issue.

No problem, I am used to it. I'd estimate around 90% of reports here on the Realtek drivers are actually distro or configuration problems. That is not to say these Realtek drivers are good, because they are not which is why recommend Mediatek adapters.

> Finally, which is the current and most compatible adapter you can recommend for pentest?

I'd say in this order:

Alfa ACHM
Alfa ACM
Alfa AXML
Alfa AXM

The ACHM supports Active Monitor and is the best for pentesting according the experts I talk to. This adapter has exceptional range.

If you need 6 GHz support then the AXM or ACML.

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was addressed via driver-binding conflict handling plus validated live hot-switch logic (native <-> out-of-tree) without reboot/reinstall.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Install-time native-driver blacklist: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/install-driver.sh#L279\n- Live bind switch tool: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/hot-switch-driver.sh#L3\n- Runtime binding diagnostics: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L29\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.