morrownr/8814au #134: add Mac timestamp support

RootLUG, 2024-05-14:
As title says, add a support for logging Mac timestamp (e.g. the value in microseconds of the MAC's 64-bit 802.11 Time Synchronization Function timer when the first bit of the MPDU arrived at the MAC). This must be provided by the driver other the field is missing/empty in the radiotap header when capturing traffic. https://www.radiotap.org/fields/TSFT.html

This would increase the capabilities as the "timestamp" of packet in pcap is not of a high accuracy and depends on load/processing speed as compared to the mac timestamp which would allow the devices to be used in more specialized cases such as multilateration of radio signal by computing time-of-flight (for which mac timestamp is needed to get high enough precision)

morrownr, 2024-05-14:
@RootLUG 

I can appreciate your suggestion but I am not going to be adding features to this driver. I hope to have time to be part of an effort later this year to bring a standards compliant driver for this chipset to the Linux kernel. I will post here if it comes to pass.

RootLUG, 2024-05-15:
Understood, great that there is a plan to make it officially supported in a linux kernel. Thank you for your hard work and contributions

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.