morrownr/8814au #135: Ubuntu 22.04.04  6.5.0-35-generic

SUTLZY, 2024-05-22:
[sudo] password for dlut:
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 64/64 (in-use/total processing units)
: 230683896 (total system memory)
: 6.5.0-35-generic (kernel version)
: gcc (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
: dkms-2.8.7
: SecureBoot disabled
: ---------------------------

Checking for previously installed drivers.
rtl8814au/5.8.5.1: added
: ---------------------------

Removing a dkms installation.
Deleting module rtl8814au-5.8.5.1 completely from the DKMS tree.
Removing 8814au.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl8814au-5.8.5.1
Removal complete.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8814au-5.8.5.1
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
kernelver=6.5.0-35-generic ./dkms-make.sh...(bad exit status: 126)
ERROR (dkms apport): binary package for rtl8814au: 5.8.5.1 not found
Error! Bad return status for module build on kernel: 6.5.0-35-generic (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 0.37 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh

Beveiligde, 2024-06-13:
Same issue here with Linux kali 6.8.11-amd64 using VirtualBox. Any solution for this? I tried removing all instalation for this driver and reboot, but still showing that error. 

Thanks in advance.

Beveiligde, 2024-06-13:
Thank you for the answer.

I managed to make it work now. Even the file 'install-driver.sh' was not executed completely (it asked always to run ./remove-install.sh, reboot, and try again to run ./install-driver.sh due error number 6 [?] ). 

What I did is check that the awus1900 was connected to the correct USB connector of my laptop. Then I went to USB configuration of my Kali Linux VM from VirtualBox, and added the driver in the USB filters. After this I run for the file ./install-driver.sh for a last time (even if I receive an "error 6" output), I ignored the error. Then, I checked that the Driver was detected by my Kali Linux VM with 'lsusb', which shows:

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 80ee:0021 VirtualBox USB Tablet

I tried to run iwconfig, and the wlan0 is showed now. To make sure it was working I tried "$~wifite --kill", and the SSIDs around me were detected by awus1900.

I am not very into network testing, but I heard of people that cannot sniff packets nor deauthenticate clients with airmon tools, so there may be more things to check or configure before all work fine. I will check more functionalities of the driver in the next days to confirm that the driver is working correctly.

If I get some type of relevant error I will let you know here to see if there is more people with the same issue, and try to find solutions.

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: DKMS now invokes `dkms-make.sh` via `sh` to avoid execute-bit/126 failures in DKMS build roots. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was mitigated by executing DKMS builds via shell invocation so environments with missing execute bits do not fail with code 126.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- DKMS MAKE command fix: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/dkms.conf#L4\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.