morrownr/8814au #138: Stopped connecting with internet

souravdas142, 2024-08-11:
Hi @morrownr ,

Thanks for your awesome contribution, for this project only till now my device keeps alive. 

OS: Arch Linux 
kernel - 6.10.2-arch1-2
driver version : master

Recently I am observing frequent network disconnection on kernel 6.10.x and then when I check the `journal` logs, I can found,

```bash
kernel: RTW: recv eapol packet 1/4
RTW: send eapol packet 2/4
wlp0s20u3: WPA: Key negotiation completed with 44:95:3b:87:50:01 [PTK=CCMP GTK=CCMP]
wlp0s20u3: CTRL-EVENT-CONNECTED - Connection to 44:95:3b:87:50:01 completed [id=0 id_str=]
wpa_supplicant[911]: bgscan simple: Failed to enable signal strength monitoring
RTW: recv eapol packet 3/4
RTW: send eapol packet 4/4
RTW: set pairwise key camid:4, addr:44:95:3b:87:50:01, kid:0, type:AES
set group key camid:5, addr:44:95:3b:87:50:01, kid:2, type:AES
```

and 

```bash
NetworkManager[450]: <info>  [1723355817.7205] manager: NetworkManager state is now CONNECTED_SITE
systemd[1]: Starting Network Manager Script Dispatcher Service...
systemd[1]: Started Network Manager Script Dispatcher Service.
wpa_supplicant[911]: wlp0s20u3: CTRL-EVENT-SCAN-FAILED ret=-1
kernel: perf: interrupt took too long (2511 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
```

for which it release the network connectivity. Everything was fine till `kernel 6.9` after updating `6.10` the problem I am observing.

Below is the full `log`:
[journal_log.log](https://github.com/user-attachments/files/16571809/journal_log.log)

souravdas142, 2024-08-12:
Hi @morrownr ,

There is a similar issue: https://github.com/morrownr/88x2bu-20210702/issues/69 but the OP never gets back, so the issue is closed.

But, I have been using it for more than 3  years, never got any problem apart from the last week to till now.

Can you please look into this ?

morrownr, 2024-08-12:
>  I am observing frequent network disconnection on kernel 6.10.x

You are not the only one seeing wifi problems with kernel 6.10. There are issues being fixed. My advice is for you to roll back to a 6.6 to 6.8 kernel for now and wait for things to be fixed.

I do not think this problem is specific to this driver and I am cautious about digging around in this driver at this point as it is old and fragile.

There should be a new rtl8814au driver added to rtw88 in the kernel over the next few months.

souravdas142, 2024-08-21:
Hey @morrownr sorry to bother you.

> There should be a new rtl8814au driver added to rtw88 in the kernel over the next few months.

This is a great news. I will be looking for it :smile: . Thanks again. but any `ETA` for this ?

And yes There may be hardware issue. because I have plugged in on old system (Arch atleast 3 months with no update) there is the same issue, even I changed the cable. Not sure Now its the device hardware issue or USB connector issue.

Any Idea for further testing ?

morrownr, 2024-08-21:
> This is a great news. I will be looking for it 😄 . Thanks again. but any ETA for this ?

No ETA yet. Right now the new driver for rtl8812au, rtl8821au and rtl8811au are in the process of going through quality control at linux-wireless. Once that is complete, the drivers will merged into wireless-next and will away going into the next kernel which is likely to be kernel 6.12.

Once that is complete, planning and work on rtl8814au can begin. 

> Any Idea for further testing ?

Not really. I watch what is going with Linux development. There is no doubt in my mind that it is the very recent Linux wifi stack that is causing this problem. There are patches going in regularly. Testing with newer kernels is a good idea when they are available. 

Since I saw your message, I have tested with Debian 12 with kernel 6.1 and Ubuntu 24.04 with kernel 6.8. I saw no problems. Using bleeding edge rolling distros comes with a risk... you are basically beta testing the latest Linux kernel. If you are in immediate need, go over to my Plug and Play List and see about getting another adapter:

The site Main Menu is:

https://github.com/morrownr/USB-WiFi

The Plug and Play List is menu item 2. A quick pick would be the EDUP EP-AX1672. More details and more adapters can be found by scroll down on the Plug and Play List. A good plug and play adapter is always handy to have around.

souravdas142, 2024-09-20:
Hi   @morrownr   Sorry to troubled you that many Days...

Today I fixed usb ports (after getting same problem when I connect another adapter from friend. I suspected its the issue of usb ports) and It got fixed.

Thanks a lot for troubleshoots.... And sorry I made a mess here

Good work... and I am happy this driver will be part of mainline kernel. :smile:

closed 2024-09-20 by souravdas142