morrownr/8814au #139: Build issue on 5.14 kernel / Rockylinux 9

CykaBlyn, 2024-08-11:
Hi there,

I am stuck trying to install RT8814AU driver for my AWUS1900 WIFI card.

Here are the informations you can use to help me :

root@localhost#sh install-driver.sh
> : ---------------------------
> : install-driver.sh v20230227
> : x86_64 (architecture)
> : 4/4 (in-use/total processing units)
> : 15889952 (total system memory)
> : 5.14.0-427.28.1.el9_4.x86_64 (kernel version)
> : gcc (GCC) 11.4.1 20231218 (Red Hat 11.4.1-3)
> : dkms-3.0.13
> : SecureBoot disabled
> : ---------------------------
> 
> Checking for previously installed drivers.
> : ---------------------------
> 
> Starting installation.
> Installing 8814au.conf to /etc/modprobe.d
> The dkms installation routines are in use.
> Copying source files to /usr/src/rtl8814au-5.8.5.1
> Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
> The driver was added to dkms successfully.
> : ---------------------------
> Sign command: /lib/modules/5.14.0-427.28.1.el9_4.x86_64/build/scripts/sign-file
> Signing key: /var/lib/dkms/mok.key
> Public certificate (MOK): /var/lib/dkms/mok.pub
> 
> Building module:
> Cleaning build area...
> Building module(s).......(bad exit status: 2)
> Failed command:
> kernelver=5.14.0-427.28.1.el9_4.x86_64 ./dkms-make.sh
> Error! Bad return status for module build on kernel: 5.14.0-427.28.1.el9_4.x86_64 (x86_64)
> Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
> An error occurred. dkms build error:  10
> Please report this error.
> Please copy all screen output and paste it into the problem report.
> Run the following before reattempting installation.
> $ sudo ./remove-driver.sh

root@localhost#cat /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log

> DKMS make.log for rtl8814au-5.8.5.1 for kernel 5.14.0-427.28.1.el9_4.x86_64 (x86_64)
> Sun Aug 11 15:53:23 CEST 2024
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.14.0-427.28.1.el9_4.x86_64/build M=/var/lib/dkms/rtl8814au/5.8.5.1/build  modules
> make[1]: Entering directory '/usr/src/kernels/5.14.0-427.28.1.el9_4.x86_64'
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_cmd.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_security.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_debug.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_io.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_query.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_set.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ieee80211.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme_ext.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mi.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_wlan_util.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_vht.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_pwrctrl.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rf.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_chplan.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_recv.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sta_mgt.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ap.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh_pathtbl.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh_hwmp.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_xmit.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_p2p.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rson.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_tdls.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_iol.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sreset.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_btcoex_wifionly.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_btcoex.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_beamforming.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_odm.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm_fsm.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm_util.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/efuse/rtw_efuse.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_intf.o
> /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.c: In function ‘thread_exit’:
> /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.c:1297:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
>  1297 |         complete_and_exit(comp, 0);
>       |         ^~~~~~~~~~~~~~~~~
>   CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_ops_linux.o
> /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.c: At top level:
> cc1: note: unrecognized command-line option ‘-Wno-enum-int-mismatch’ may have been intended to silence earlier diagnostics
> cc1: some warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:299: /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c: In function ‘rtw_os_ndev_register’:
> /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:1907:9: error: too many arguments to function ‘netif_napi_add’
>  1907 |         netif_napi_add(ndev, &adapter->napi, rtw_recv_napi_poll, RTL_NAPI_WEIGHT);
>       |         ^~~~~~~~~~~~~~
> In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/osdep_service_linux.h:31,
>                  from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/osdep_service.h:57,
>                  from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:27,
>                  from /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c:17:
> ./include/linux/netdevice.h:2732:1: note: declared here
>  2732 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
>       | ^~~~~~~~~~~~~~
> /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.c: At top level:
> cc1: note: unrecognized command-line option ‘-Wno-enum-int-mismatch’ may have been intended to silence earlier diagnostics
> make[2]: *** [scripts/Makefile.build:299: /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o] Error 1
> make[1]: *** [Makefile:1936: /var/lib/dkms/rtl8814au/5.8.5.1/build] Error 2
> make[1]: Leaving directory '/usr/src/kernels/5.14.0-427.28.1.el9_4.x86_64'
> make: *** [Makefile:2395: modules] Error 2

uname -a
> Linux localhost 5.14.0-427.28.1.el9_4.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Jul 31 15:28:35 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

root@localhost#cat /etc/rocky-release

> Rocky Linux release 9.4 (Blue Onyx)

Let me know if you need more informations !

Thanks for your help.

morrownr, 2024-08-11:
Hi @CykaBlyn 

Unfortunately Rocky is based on RHEL and RHEL presents special problems in cases like this because API patches are selectively moved from where they are in mainline. I do not support RHEL because I don't have the time but I allow RHEL users to add code to these drivers if they show me they know what they are doing. The only driver here at this site that is ready for Rocky is:

https://github.com/morrownr/88x2bu-20210702

You do have some alternatives. We would need to look and see what drivers and firmware is included in Rocky but often drivers and firmware for the in-kernel drivers based on the mt7610u, mt7612u and mt7921au chips are included so it would be plug and Play. We would need to check and we can do that if you are interested.

CykaBlyn, 2024-08-14:
This is the reply i was expecting, sadly. 

I figured it out by using another OS (Ubuntu 24). Thanks for your reply ! If you need any help from a RHEL user (testing mostly), dont hesitate to ask :).

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: RHEL9 backport compatibility guards added for `complete_and_exit` and NAPI API drift. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was mitigated with compatibility guards for RHEL-family kernel API drift across thread-exit, NAPI registration, and cfg80211 management registration paths.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Thread-exit compatibility guard: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/os_dep/osdep_service.c#L1294\n- NAPI compatibility guard: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/os_dep/linux/os_intfs.c#L1911\n- cfg80211 mgmt registration compatibility: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/os_dep/linux/ioctl_cfg80211.c#L10034\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.