morrownr/8814au #140: dkms build error:  10

interconnectedMe, 2024-08-13:
I had an issue after updating KDE Neon today.

so I tried to follow the upgrade instructions here on github, that did not seem to work.

Then I tried uninstalling, rebooting, and installing again, but got an error to the effect that 'directory already exists and is not empty'

Then I tried uninstalling again.  (I'm not savvy enough with gnu/linux to find the directory and delete it manually after running the uninstall script.

I noted the prompt to paste the terminal screen contents in to the problem report, so they are pasted below.  I think I've managed to attach 'make.log' also if that helps.

I have no idea what to do after this, so if anyone knows, please do let me know ;-)

`
darren@drneon:~/src/8814au$ sudo ./install-driver.sh
[sudo] password for darren: 
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 8/8 (in-use/total processing units)
: 16250604 (total system memory)
: 6.8.0-40-generic (kernel version)
: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
: dkms-2.8.7
EFI variables are not supported on this system
EFI variables are not supported on this system
EFI variables are not supported on this system
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8814au-5.8.5.1
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
The driver was added to dkms successfully.
: ---------------------------

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
./dkms-make.sh......(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.8.0-40-generic (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 59.00 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
`
[make.log](https://github.com/user-attachments/files/16604067/make.log)

interconnectedMe, 2024-08-13:
Something worked.
I think previously 'git pull' did not work (at least I don't recall seeing the output I saw this time.
This time, 'git pull' read back a lot of 'stuff' (40/40, Delta 22 iirc)
Wifi now working ridiculously well.  4 times-ish faster than before the updates I ran on KDE Neon today.

Thanks very much again!

morrownr, 2024-08-13:
Glad you figured it out as that would have been my first suggestion. These drivers have to be kept up to date with `git pull`.

FYI: There may be a new driver for this chip in the next few months. It will be going into mainline as part of rtw88. I will be posting information about testing here at this repo when it is available for testing.

xecutable, 2025-02-11:
I'm having the same issue with the latest CachyOs. The error reads that Clang 19 was used, but I'm trying to compile with gcc 14. Everything is up to date, so I'm not sure why it wont let me install

morrownr, 2025-02-11:
Hi @xecutable 

I can set aside some time to look at this later in the week but there is another solution these days.

https://github.com/lwfinger/rtw88

That is where we have been working on the new driver that is going into the kernel. The process of this new rtw88_8814au driver going into the kernel is underway so it should be available in kernel 6.15 or 6.16. In the meantime you can install it at the website above. We can use test reports. Remember to uninstall this driver:

$ sudo sh remove-driver.sh

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: Arch validation: clean DKMS remove/install now removes stale versions and installs current version successfully. This included a direct code-path fix and validation in the current maintenance cycle.