morrownr/8814au #141: Doesnt work on Arch Linux Kernel 6.10.6 zen

Zer0Meow, 2024-08-23:
It did use to work on older zen versions and i also have the headers and the wifi adapter is being seen but doesnt go into monitor mode or doesnt connect to anything at all.

morrownr, 2024-08-23:
@Zer0Meow 

There are a lot of issues going on with the wifi stack right now. Recommend you go back to kernel 6.6 for now if that is available.

noelbautista91, 2024-10-04:
With kernel 6.6 I have to keep rebuilding this repo every time I run an update. If anyone has a solution to this I am all ears.

morrownr, 2024-10-04:
Hi @noelbautista91 

Are you using this repo or a version of it that has been pulled into the AUR? If it is AUR, you need to ask whoever supports stuff in the AUR. If you are fully using this repo and are following the installation steps in the README, I 'll be glad to help.

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: added install-time blacklist for in-kernel `rtw88_8814au` to prevent USB ID binding conflicts with out-of-tree `8814au`; added install/remove runtime conflict warnings; added snapshot-backed hot-switch script to rebind without recompile/uninstall/reboot; validated both native<->oot switch directions on sx1. This included a direct code-path fix and validation in the current maintenance cycle.