morrownr/8814au #142: Can't install on openSUSE 15.6 Leap w/ Secure Boot On

Joseph-Seed-18, 2024-08-25:
I tried to install the driver on openSUSE, but it ended up outputting this error. I believe it might be because I couldn't check if 'sign-file' (for Secure Boot) was installed because I couldn't find it, or any equivalent in the openSUSE repos. The official 8814au driver for openSUSE (https://software.opensuse.org//download.html?project=hardware&package=rtl8814au) doesn't work for me anymore, since I decided to start using Secure Boot on my system. Could I get some help with installation? I really need this driver.

Make log is attached below:

`DKMS make.log for rtl8814au-5.8.5.1 for kernel 6.4.0-150600.23.17-default (x86_64)
Sat Aug 24 10:39:23 PM CDT 2024
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.4.0-150600.23.17-default/build M=/var/lib/dkms/rtl8814au/5.8.5.1/build  modules
make[1]: Entering directory '/usr/src/linux-6.4.0-150600.23.17-obj/x86_64/default'
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_btcoex_wifionly.o
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:9954:19: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_beacon = cfg80211_rtw_change_beacon,
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:9954:19: note: (near initialization for ‘rtw_cfg80211_ops.change_beacon’)
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_btcoex.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_mcc.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_hci/hal_usb.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/led/hal_led.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/led/hal_usb_led.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/HalPwrSeqCmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/rtl8814a/Hal8814PwrSeq.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/rtl8814a/rtl8814a_xmit.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/rtl8814a/rtl8814a_sreset.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/rtl8814a/rtl8814a_hal_init.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/rtl8814a/rtl8814a_phycfg.o
cc1: warning: unrecognized command line option ‘-Wno-enum-conversion’
cc1: warning: unrecognized command line option ‘-Wno-stringop-overread’
cc1: warning: unrecognized command line option ‘-Wno-enum-int-mismatch’
cc1: warning: unrecognized command line option ‘-Wno-cast-function-type’
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-6.4.0-150600.23.17/scripts/Makefile.build:252: /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [../../../linux-6.4.0-150600.23.17/Makefile:2066: /var/lib/dkms/rtl8814au/5.8.5.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-6.4.0-150600.23.17-obj/x86_64/default'
make: *** [Makefile:2395: modules] Error 2`

morrownr, 2024-08-25:
Hi @Joseph-Seed-18 

It appears some things are blowing up before you get to secure boot support. That 6.4 kernel is interesting. It is not an LTS kernel and is no longer supported and is dated. I generally hesitate to help with problems on systems that do not have a supported kernel. Do you have the option of moving to a more modern kernel such as 6.6?

Also, can you run `sudo sh install-driver.sh` again and post the output here.

Joseph-Seed-18, 2024-08-25:
Thanks for your response. I didn't realize the Leap repos supplied such an outdated kernel, as I hadn't received any kernel updates recently; I will try to see if I can update it or backport a newer kernel, and then run the script again.

Joseph-Seed-18, 2024-08-26:
Unfortunately, I wasn't able to backport a newer kernel for my openSUSE 15.6 Leap installation. However, I upgraded to openSUSE Tumbleweed, which provides a stable version of kernel 6.10. After running the install script and manually enrolling the MOK key with the command on the README page (the system didn't give me the prompt to enroll the MOK when rebooting the first time), it works perfectly now. 

Sorry I couldn't be of anymore help, but I just didn't have the time to go through all of that troubleshooting. I would probably advise users against using Leap 15.6 or earlier, considering that they supply outdated dated kernels.

closed 2024-08-26 by Joseph-Seed-18

VortexAcherontic, 2024-08-26:
> Unfortunately, I wasn't able to backport a newer kernel for my openSUSE 15.6 Leap installation. However, I upgraded to openSUSE Tumbleweed, which provides a stable version of kernel 6.10. After running the install script and manually enrolling the MOK key with the command on the README page (the system didn't give me the prompt to enroll the MOK when rebooting the first time), it works perfectly now.
> 
> Sorry I couldn't be of anymore help, but I just didn't have the time to go through all of that troubleshooting. I would probably advise users against using Leap 15.6 or earlier, considering that they supply outdated dated kernels. Thank you for your help, I really appreciate it.

openSUSE driver package maintainer here.

It kind of surprises me as the driver builds just fine on OBS for Leap. I guess I'll just try and enable signing the driver for secure boot for openSUSE then for future users. 😅

VortexAcherontic, 2024-08-26:
For the record and since there is no dedicated bugzilla ticket for openSUSE:

Enabled driver singing for the openSUSE package for Leap and Tumbleweed. I briefly tested it with Kalpa (Tumbleweed based) on a secure boot enabled system and the driver seems to loads and detect supported devices properly.

Tested with an ASUS USB-AC68 wireless USB adapter.

joseguzman1337, 2026-02-12:
Hi @ISAAK250,

Thanks for following up with the outcome.

**OS/environment (from your report)**
- openSUSE Leap 15.6 (kernel `6.4.0-150600.23.17-default`) with Secure Boot enabled
- Upgrade path that worked for you: openSUSE Tumbleweed (kernel `6.10`), then install + MOK enrollment

For Secure Boot systems, the driver can build/install but still fail to load or create an interface until the module signature is accepted (MOK enrollment). The key point from your report is that on Tumbleweed it worked once the MOK was explicitly enrolled.

For anyone troubleshooting this on openSUSE, please update/upgrade first (OS packages, kernel, this driver, and firmware stack where applicable: BIOS/UEFI + device firmware), then collect:
```sh
cat /etc/os-release
uname -a
mokutil --sb-state || true
mokutil --list-enrolled || true
modinfo 8814au | sed -n '1,120p'
modprobe -n -v 8814au || true
sudo dmesg -T | tail -n 250
```

Also, thanks to @VortexAcherontic (openSUSE package maintainer) for enabling module signing in the openSUSE package, which should improve the Secure Boot experience for future users.