morrownr/8814au #148: install-driver.sh fails when SecureBoot is enabled and DKMS is not used

thonigmann, 2024-10-13:
This is Ubuntu 24.04 running in a virtual machine, but I'm sure this will occur on real hardware as well.
Secure Boot is enabled and DKMS is not installed.
Calling sudo ./install-driver.sh fails with a message "make: sign: File not found"

Apparently, the problem is the definition of the target _sign-install_ in Makefile, which orders make to call a program called sign that doesn't exist.
```
sign-install:
        sign install
```
Changing the definition to let the target _sign-install_ forward to the targets _sign_ and _install_ fixes the issue.
``` 
sign-install: sign install
```
Here is the complete output of install-driver.sh:

> thomas@smimas:~/src/8814au$ sudo ./install-driver.sh 
> : ---------------------------
> : install-driver.sh v20230227
> : x86_64 (architecture)
> : 4/4 (in-use/total processing units)
> : 3960492 (total system memory)
> : 6.8.0-45-generic (kernel version)
> : gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
> : SecureBoot enabled
> : ---------------------------
> 
> Checking for previously installed drivers.
> : ---------------------------
> 
> Starting installation.
> Installing 8814au.conf to /etc/modprobe.d
> The non-dkms installation routines are in use.
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.8.0-45-generic/build M=/home/thomas/src/8814au  modules
> make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-45-generic“ wird betreten
> warning: the compiler differs from the one used to build the kernel
>   The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
>   You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
>   CC [M]  /home/thomas/src/8814au/core/rtw_cmd.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_security.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_debug.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_io.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_ioctl_query.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_ioctl_set.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_ieee80211.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_mlme.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_mlme_ext.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_mi.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_wlan_util.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_vht.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_pwrctrl.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_rf.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_chplan.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_recv.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_sta_mgt.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_ap.o
>   CC [M]  /home/thomas/src/8814au/core/mesh/rtw_mesh.o
>   CC [M]  /home/thomas/src/8814au/core/mesh/rtw_mesh_pathtbl.o
>   CC [M]  /home/thomas/src/8814au/core/mesh/rtw_mesh_hwmp.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_xmit.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_p2p.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_rson.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_tdls.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_br_ext.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_iol.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_sreset.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_btcoex_wifionly.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_btcoex.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_beamforming.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_odm.o
>   CC [M]  /home/thomas/src/8814au/core/rtw_rm.o
[... 91 lines cut ...]
>   CC [M]  /home/thomas/src/8814au/hal/phydm/rtl8814a/halhwimg8814a_mac.o
>   CC [M]  /home/thomas/src/8814au/hal/phydm/halrf/rtl8814a/halhwimg8814a_rf.o
>   CC [M]  /home/thomas/src/8814au/hal/phydm/halrf/rtl8814a/halrf_iqk_8814a.o
>   CC [M]  /home/thomas/src/8814au/hal/phydm/rtl8814a/phydm_regconfig8814a.o
>   CC [M]  /home/thomas/src/8814au/hal/phydm/halrf/rtl8814a/halrf_8814a_ce.o
>   CC [M]  /home/thomas/src/8814au/hal/phydm/rtl8814a/phydm_rtl8814a.o
>   CC [M]  /home/thomas/src/8814au/hal/phydm/txbf/haltxbf8814a.o
>   CC [M]  /home/thomas/src/8814au/platform/platform_ops.o
>   LD [M]  /home/thomas/src/8814au/8814au.o
>   MODPOST /home/thomas/src/8814au/Module.symvers
>   CC [M]  /home/thomas/src/8814au/8814au.mod.o
>   LD [M]  /home/thomas/src/8814au/8814au.ko
>   BTF [M] /home/thomas/src/8814au/8814au.ko
> Skipping BTF generation for /home/thomas/src/8814au/8814au.ko due to unavailability of vmlinux
> make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-45-generic“ wird verlassen
> : SecureBoot enabled - read FAQ about SecureBoot
> sign install
> make: sign: Datei oder Verzeichnis nicht gefunden
> make: *** [Makefile:2414: sign-install] Fehler 127
> An error occurred:  2
> Please report this error.
> Please copy all screen output and paste it into the problem report.
> You will need to run the following before reattempting installation.
> $ sudo ./remove-driver.sh
>

morrownr, 2024-10-13:
@thonigmann 

Thanks. This error is in all the repos here. I am working on fixing this and another bug I found in that fork on the code.

It is a lot of work maintaining several out-of-kernel drivers but the good news is that 4 of the ones I maintain here should have drivers in the kernel that are of good enough quality that they are better than the out-of-kernel drivers here so I should be able to look at archiving some of the drivers here in 2025.

The drivers in the rtw88 kernel driver are 88x2bu, 8821cu and the quality has been improving greatly over this year. If you are using kernel 6.11 or later, you should not need these drivers anymore. Soon to be added are drivers for 8812au and 8821au. These will be very good drivers and should go into kernel 6.13. A rtl8814au driver is planned and can maybe happen sometime in 2025. This is all community work... the Realtek usb wifi team is not involved so please don't give them any credit. Buy Mediatek based adapters if you care about good Linux drivers in usb adapters.

thonigmann, 2024-10-20:
Thank you for the info and thank you for the work you do. It's not an urgent issue for me since I know how to fix it locally.