morrownr/8814au #149: Raspberry Pi5 8Gb / kali-linux-2024.3-raspberry-pi-arm64

raffak024, 2024-11-14:
Anyone got it to working?

morrownr, 2024-11-14:
What seems to be the problem?

Well, even if you get it going, since you are using Kali. you likely will not be happy. Let me suggest:

The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Reading menu item 1 is a good idea and menu item 2 has The Plug and Play List and under that is a Recommended Adapters for Kali Linux list.

Monitor mode is not something Realtek does very well. You really need to move to an adapter with a Mediatek chipset.

raffak024, 2024-11-14:
Shit... I waited two months for this Adapter... So then... 4 days of searching to get it working. For now I give up.

What is your favourite Adapter? :)

-------- Ursprüngliche Nachricht --------

morrownr, 2024-11-14:
> Shit...

Not the first time I have heard this. This is why I created my USB-WiFi site. Linux USB WiFi support is better than even and rapidly getting better but you need to know what to buy. Maybe someday that won't be an issue but for now, it is.

> What is your favourite Adapter? :)

I have several favorite adapters. It depends on what I am doing. For monitor mode work with Kali, I would say the Alfa ACHM is my favorite. There are others I would certainly use for monitor mode work. You would have to tell me what you plan to do. The more detail, the better. The ACHM uses a mt7610u chip and the in-kernel driver is very feature rich. The adapter is plug and play with Linux. There is also the Alfa ACM. It uses the mt7612u chip and again, the driver is in-kernel and is feature rich. I have an EDUP EP-AX1672 adapter that uses the mt7921au chip and mt7921u in-kernel driver. You can get more info from the Kali recommended list:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_Adapters_for_Kali_Linux.md

On the Main Menu, link is in my first reply, The Plug and Play List shows the recommended Kali adapters plus many more in a lot of detail.

Cheers

ToySldr, 2024-12-22:
Same thing happened to me! I bought this epic WiFi card to use with my Pi5 and, nothing seems to be working properly. Time to try something different.

morrownr, 2024-12-22:
@ToySldr @raffak024 

The rtl8814au chipset has never been supported in Linux very well... until now. Two days ago a new rtw8814au driver was released for testing. You can go to the following site to install it for testing:

https://github.com/lwfinger/rtw88

I work at that site also. The idea is for that new driver to go into the Linux kernel. The instructions there are different than here so remember to uninstall this driver:

sudo sh remove-driver.sh

And follow the instructions at the new site. We really do need reports good or bad.

I still recommend looking for adapters on the Main Menu here at this site if you are shopping for adapters:

https://github.com/morrownr/USB-WiFi

raffak024, 2024-12-23:
Got the AWUS1900 working. Monitoring Mode and packet injection.

raffak024, 2024-12-23:
> @ToySldr @raffak024
>
> The rtl8814au chipset has never been supported in Linux very well... until now. Two days ago a new rtw8814au driver was released for testing. You can go to the following site [...]

Got the AWUS1900 working. Monitoring Mode and packet injection. <3

morrownr, 2024-12-23:
> Got the AWUS1900 working. Monitoring Mode and packet injection. <3

That is good to hear. We need positive reports as well as bug reports over at the repo where you got the driver. We are looking at early Feb. as a target to go upstream to the kernel with it. So not much time to test and fix. Test the heck out of it and spread the word to anyone else you know that has a 8814au based adapter.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was addressed via driver-binding conflict handling plus validated live hot-switch logic (native <-> out-of-tree) without reboot/reinstall.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Install-time native-driver blacklist: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/install-driver.sh#L279\n- Live bind switch tool: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/hot-switch-driver.sh#L3\n- Runtime binding diagnostics: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L29\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.