morrownr/8814au #150: Not able to install on Linux Mint 22(Kernel: 6.8.0-48-generic)

Mehmetakif2, 2024-11-17:
Hi a newbie here. I recently formated my pc and wanted to reinstall the driver i used before formating my pc but I am having troubles right now. I tried to install from aircrack but there i couldnt even compile it because of an error related to ipx.h. So i tried this one which was installed on my pc before the reset. Here i can compile the driver but I am unfortunately not able to use my wifi adapter after reboot. The driver is shown in the dkms status as rtl8814au/5.8.1.5 but can not use the device.
I would appreciate it if you helped me.
Thank you

morrownr, 2024-11-17:
Hi @Mehmetakif2 

I need some info:

Paste the screen output provided by the install-driver.sh script.
The name and version of the Linux distro you are using.
The information obtained with the below command.

sudo lsusb; rfkill list all; dkms status; iw dev; iw reg get

Mehmetakif2, 2024-11-18:
The Output of the install-driver.sh script:
: ---------------------------
: install-driver.sh v20230227
: x86_64 (architecture)
: 16/16 (in-use/total processing units)
: 65762964 (total system memory)
: 6.8.0-48-generic (kernel version)
: gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0
: dkms-3.0.11
: SecureBoot enabled
: ---------------------------

Checking for previously installed drivers.
: ---------------------------

Starting installation.
Installing 8814au.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl8814au-5.8.5.1
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
The driver was added to dkms successfully.
: ---------------------------
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
kernelver=6.8.0-48-generic ./dkms-make.sh......
Signing module /var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.ko
Cleaning build area...
Compile time: 125.33 seconds
The driver was built by dkms successfully.
: ---------------------------

8814au.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-48-generic/updates/dkms/
depmod...
The driver was installed by dkms successfully.
: ---------------------------

Info: Upgrade this driver with the following commands as needed:
$ git pull
$ sudo sh install-driver.sh
Note: Upgrades to this driver should be performed before distro upgrades.
Note: Upgrades can be performed as often as you like.
Note: Work on this driver is continuous.
: ---------------------------

I am using Linux Mint 22 Cinnamon right now with the kernel 6.8.0-48-generic

The output of sudo lsusb:
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 046d:c33f Logitech, Inc. G815 Mechanical Keyboard
Bus 001 Device 004: ID 14cd:125d Super Top Mass Storage Device
Bus 001 Device 005: ID 1462:7c56 Micro Star International MYSTIC LIGHT 
Bus 001 Device 006: ID 046d:c08b Logitech, Inc. G502 SE HERO Gaming Mouse
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0846:9054 NetGear, Inc. Nighthawk A7000 802.11ac Wireless Adapter AC1900 [Realtek 8814AU]
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

There is unfortunately no output for the command rfkill list all.

The output of dkms status: rtl8814au/5.8.5.1, 6.8.0-48-generic, x86_64: installed

There is unfortunately no output for iw dev either.

The output of iw reg get:
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

morrownr, 2024-11-18:
Hi @Mehmetakif2 

> : SecureBoot enabled

This likely has something to do with the problem. To test, you could turn secureboot off in the bios and reboot.

Mehmetakif2, 2024-11-18:
Oh thank you i solved the problem with another solution. I was able to use the driver with secure boot because i used a MOK key to sign it. This time i couldnt find the driver so i tought that this isnt possible anymore but i found the compressed file in /lib/modules so i could decompress it and sign it. Now i can use it perfectly with secure boot enabled.
Thank you very much morrownr. I appriciate your work.

closed 2024-11-18 by Mehmetakif2

morrownr, 2024-11-18:
Glad you you got it going. Secureboot is one of those things where every distros seems to do something differently. The install script generally works well with Ubuntu and those systems built on Ubuntu like Mint but there can be other issues as well. Hard to get this right.

For your information: The plan is to start working on an in-kernel version of a rtl8814au driver soon. I will post a message at the top of the readme.md when it is ready for testing. The work will be done in this repo:

https://github.com/lwfinger/rtw88