morrownr/8814au #151: WPA3

p1r473, 2024-11-21:
Hi, 
Alfa has WPA3 working in their drivers. Is it possible to implement that into your drivers? I don't see any mention of it's this driver supporting WPA3 on the readme. 

Thanks

https://store.rokland.com/blogs/news/wpa3-security-support-now-available-for-awus036ac-awus036ach-awus1900

morrownr, 2024-11-21:
Hi @p1r473 

Our best bet for WPA3 support is the plan to develop a new driver and push it into the Linux kernel. This week, new drivers for the rtl8812au and rtl8821/11au chip went into kernel 6.13. The plan is to work on rtl8814au next. Hopefully something will be ready for testing in the next few months. You can keep a watch here as I will post at the top of the README when it is ready for testing.

closed 2024-11-22 by p1r473

p1r473, 2024-11-22:
@morrownr to clarify, it is you that is adding the support, or alfa? So its for sure on your roadmap? (considering if I am returning the adapter if it doesnt get WPA3)

morrownr, 2024-11-22:
@p1r473 

> So its for sure on your roadmap?

There are no guarantees. Plans don't always work out so if you can return the adapter, I recommend that you so and tell the company that sold it to you that Linux driver support is inadequate.

Here is the Main Menu for my github site:

https://github.com/morrownr/USB-WiFi

Recommend you read menu item 1 to get an overall idea of the USB WiFi Adapter situation for Linux. Then you can take a look at the items on menu item 2. The Plug and Play List is a really good source of adapters that are known to work well with Linux. There is also a separate list if you use Kali. You are welcome to ask all of the questions you have.

There is no need to go through all of the pain that Realtek based adapters can inflict. Realtek USB WiFi driver support is bad and has been for a long time. Mediatek support is really good.

joseguzman1337, 2026-02-12:
Hi @p1r473,

From your logs/config this looks like **AP/Hotspot mode** via NetworkManager (`mode=ap`) rather than a normal WPA3 client/STA connection.

A couple of clarifications that usually matter for WPA3 troubleshooting:
- `iw list` reporting `Device supports SAE with AUTHENTICATE command` indicates **SAE capability exists in the stack**, but it does not guarantee that **AP mode with WPA3-SAE** is supported/working in a given driver + userspace combination.
- For WPA3 *hotspot/AP*, the usual working path is `hostapd` with SAE (nl80211), and it requires driver support for AP mode + the relevant key management operations.

**Next steps (please retest with an updated baseline)**
Before we dig deeper, please update/upgrade to the latest available versions for:
- OS packages (full update)
- kernel
- this driver (latest `main`)
- system firmware stack where applicable (BIOS/UEFI + device firmware)

Then, please share the following outputs so we can tell whether this is a driver limitation, a userspace config issue, or something else:
```sh
cat /etc/os-release
uname -a
nmcli --version
hostapd -v || true
wpa_supplicant -v || true
iw --version
iw list
iw dev
rfkill list
modinfo 8814au | sed -n '1,80p'
sudo dmesg -T | tail -n 250
journalctl -u NetworkManager -b --no-pager | tail -n 250
```

Also confirm:
- Are you trying to create a WPA3 hotspot (AP) or connect to an existing WPA3 network (client/STA)?
- If hotspot/AP: are you using NetworkManager’s hotspot, or `hostapd` directly?

Longer-term: as @morrownr noted in this thread, the mac80211-based rtl8814au work in `rtw88` is the direction for better feature support (including WPA3-related behavior), but this legacy out-of-kernel driver has known limitations.