morrownr/8814au #152: 8814au: segfault at 8814au/core/rtw_mlme_ext.c:12187 rtw_mlmeext_disconnect+0x344/0x440 when reloading driver

sardChen, 2024-11-22:
When  reloading  8814au, I found a segfault bug in dmesg logs:

```
[ 2077.989802] ------------[ cut here ]------------
[ 2077.989811] WARNING: CPU: 6 PID: 7817 at /home/sandy/workplace/wifiTool/driver/8814/8814au/core/rtw_mlme_ext.c:12187 rtw_mlmeext_disconnect+0x344/0x440 [8814au]
[ 2077.989942] Modules linked in: 8814au(OE) tcp_diag inet_diag rfcomm ccm cmac algif_hash algif_skcipher af_alg bnep binfmt_misc nls_iso8859_1 amdgpu snd_sof_pci_intel_tgl snd_sof_intel_hda_common soundwire_intel snd_sof_intel_hda_mlink soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi soundwire_generic_allocation soundwire_bus snd_soc_core iwlmvm snd_hda_codec_realtek snd_compress ac97_bus snd_pcm_dmaengine snd_hda_codec_generic intel_rapl_msr ledtrig_audio intel_rapl_common intel_uncore_frequency snd_hda_codec_hdmi mac80211 intel_uncore_frequency_common snd_hda_intel x86_pkg_temp_thermal snd_intel_dspcfg snd_intel_sdw_acpi intel_powerclamp coretemp snd_hda_codec snd_hda_core kvm_intel snd_hwdep drm_exec amdxcp snd_pcm drm_buddy gpu_sched drm_suballoc_helper drm_ttm_helper pmt_telemetry libarc4 pmt_class snd_seq_midi snd_seq_midi_event ttm snd_rawmidi kvm btusb snd_seq btrtl btintel btbcm drm_display_helper
[ 2077.990044]  iwlwifi btmtk irqbypass cec bluetooth joydev crct10dif_pclmul input_leds rc_core polyval_clmulni polyval_generic i2c_algo_bit cfg80211 ghash_clmulni_intel sha256_ssse3 sha1_ssse3 snd_seq_device ecdh_generic aesni_intel ecc crypto_simd snd_timer snd cryptd cmdlinepart spi_nor soundcore mei_me rapl mei mtd ee1004 intel_cstate intel_vsec intel_hid sparse_keymap mac_hid gigabyte_wmi wmi_bmof acpi_pad acpi_tad intel_pmc_core sch_fq_codel ipmi_devintf ipmi_msghandler msr parport_pc ppdev lp parport pstore_blk ramoops pstore_zone reed_solomon efi_pstore ip_tables x_tables autofs4 hid_generic usbhid hid r8169 crc32_pclmul nvme i2c_i801 ahci intel_lpss_pci spi_intel_pci spi_intel i2c_smbus intel_lpss xhci_pci realtek nvme_core libahci idma64 xhci_pci_renesas video wmi pinctrl_alderlake
[ 2077.990155] CPU: 6 PID: 7817 Comm: RTW_CMD_THREAD Tainted: G           OE      6.6.58 #1
[ 2077.990160] Hardware name: Gigabyte Technology Co., Ltd. B660M GAMING AC DDR4/B660M GAMING AC DDR4, BIOS F4 01/17/2022
[ 2077.990163] RIP: 0010:rtw_mlmeext_disconnect+0x344/0x440 [8814au]
[ 2077.990265] Code: c1 48 c1 e9 03 48 ba 00 00 00 00 00 fc ff df 0f b6 14 11 83 e0 07 83 c0 03 38 d0 7c 04 84 d2 75 10 83 3d ce 79 2c 00 03 77 15 <0f> 0b e9 88 fd ff ff 48 c7 c7 00 c1 97 c5 e8 29 41 28 d9 eb e2 44
[ 2077.990269] RSP: 0018:ffff88828436fcf0 EFLAGS: 00010293
[ 2077.990273] RAX: 0000000000000003 RBX: ffffc90001762000 RCX: 1ffffffff8b2f820
[ 2077.990275] RDX: 0000000000000004 RSI: ffff88817ea78a00 RDI: ffffc90001762000
[ 2077.990278] RBP: ffff88828436fdb8 R08: 0000000000000000 R09: 0000000000000000
[ 2077.990280] R10: 0000000000000000 R11: 0000000000000000 R12: ffff88828436fd90
[ 2077.990282] R13: ffffc90001762024 R14: 1ffff1105086df9e R15: 0000000080000001
[ 2077.990284] FS:  0000000000000000(0000) GS:ffff888856300000(0000) knlGS:0000000000000000
[ 2077.990287] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2077.990290] CR2: 00007d82e041ff28 CR3: 000000010f1ba000 CR4: 0000000000f50ee0
[ 2077.990292] PKRU: 55555554
[ 2077.990294] Call Trace:
[ 2077.990296]  <TASK>
[ 2077.990298]  ? show_regs+0x72/0x90
[ 2077.990306]  ? __warn+0xcf/0x270
[ 2077.990310]  ? __pfx_usb_write32+0x10/0x10 [8814au]
[ 2077.990446]  ? rtw_mlmeext_disconnect+0x344/0x440 [8814au]
[ 2077.990544]  ? report_bug+0x2bf/0x320
[ 2077.990550]  ? handle_bug+0x46/0x90
[ 2077.990554]  ? exc_invalid_op+0x19/0x50
[ 2077.990557]  ? asm_exc_invalid_op+0x1b/0x20
[ 2077.990561]  ? rtw_mlmeext_disconnect+0x344/0x440 [8814au]
[ 2077.990665]  ? __pfx_rtw_mlmeext_disconnect+0x10/0x10 [8814au]
[ 2077.990763]  ? __down_common+0x221/0x540
[ 2077.990767]  ? __kasan_check_write+0x14/0x30
[ 2077.990772]  disconnect_hdl+0x117/0x210 [8814au]
[ 2077.990861]  ? __pfx_disconnect_hdl+0x10/0x10 [8814au]
[ 2077.990948]  ? __pfx_disconnect_hdl+0x10/0x10 [8814au]
[ 2077.991084]  rtw_cmd_thread+0xb93/0x10c0 [8814au]
[ 2077.991490]  kthread+0x2b9/0x390
[ 2077.991512]  ? __pfx_rtw_cmd_thread+0x10/0x10 [8814au]
[ 2077.992017]  ? __pfx_kthread+0x10/0x10
[ 2077.992038]  ret_from_fork+0x43/0x90
[ 2077.992059]  ? __pfx_kthread+0x10/0x10
[ 2077.992078]  ret_from_fork_asm+0x1b/0x30
[ 2077.992101]  </TASK>
[ 2077.992110] ---[ end trace 0000000000000000 ]---
[ 2084.481627] RTW: set bssid:00:00:00:00:00:00

```

morrownr, 2024-12-22:
@sardChen 

Sorry for the slow reply. I recommend that you see issue #153 . It is a new driver that is much better than this one. It is open for testing. Please us it and provide reports.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. This was fixed by replacing an unsafe disconnect-state path with a safe fallback action during teardown/reload races.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Safe fallback in disconnect path: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/core/rtw_mlme_ext.c#L12185\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.