morrownr/8814au #153: Project: Add 8814au in-kernel driver to the Linux Mainline kernel. Need testers...

morrownr, 2024-12-20:
Greetings to anyone that reads this message.

This Issue is to let you know that a new mac80211 compatible rtl8814au driver is available for testing. The plan is to upstream it to the Linux Mainline kernel when testing shows the driver has stability.

The driver is available for testing at the following repo:

https://github.com/lwfinger/rtw88

Remember to first remove the out-of-kernel driver in this repo or whatever repo you may have installed. You can run the following to remove it if using this repo:

$ sudo sh remove-driver.sh

It is important to follow the instructions in the README at the repo with the new test driver. You may not be familiar with rtw88 in the kernel but even if you are, there are some necessary mods that you need to know about. The rtw88 that has this new driver is more advanced than the rtw88 in stable kernels as it follows wireless-next and is used to work on and develop new drivers.

We welcome you to test and report on this new driver. Your testing will help us get this new driver in the Linux kernel sooner and in better shape.

If you are aware of anyone who is familiar with programming mac80211 drivers, please invite them as more eyes on the code is a good thing. Your ideas are most welcome. We can do this.

Issues can be posted at the rtw88 repo, where the new driver is located. The plan is to discontinue future maintenance on this driver once the new driver is in the Linux kernel as it will be a much better driver than this old driver that Realtek has not maintained very well.

@morrownr

D357R0Y3R, 2024-12-22:
Hey, @morrownr!

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapter_out-of-kernel_drivers_for_Linux.md#chipset---rtl8814au---ac1900---usb-3

For an example there are so many not supported options from the older driver, Did you managed to try to get them all working with this new driver?

morrownr, 2025-05-26:
@D357R0Y3R 

> For an example there are so many not supported options from the older driver, Did you managed to try to get them all working with this new driver?

The new driver supports many more options and features. Test it and report any features that you find missing:

https://github.com/lwfinger/rtw88