morrownr/8814au #158: rtw_switch_usb_mode=1 causes the adapter to auto-disconnect

Zerogoki00, 2025-05-10:
Hello, my adapter works fine with rtw_switch_usb_mode=0 option
I tried rtw_switch_usb_mode=1 and this cause this

`[   46.024356] usb 1-6: new high-speed USB device number 14 using xhci_hcd
[   46.172802] usb 1-6: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
[   46.172816] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   46.172823] usb 1-6: Product: 802.11ac NIC
[   46.172828] usb 1-6: Manufacturer: Realtek
[   46.172833] usb 1-6: SerialNumber: 123456
[   46.288322] RTW: HW EFUSE
[   46.288335] RTW: 0x000: 29 81 0D F4  0D 0C 18 00  80 00 00 08  08 19 12 10  
[   46.288376] RTW: 0x010: 20 20 20 20  20 20 20 20  20 20 20 00  00 00 00 EE  
[   46.288415] RTW: 0x020: EE EE 1A 1A  1A 1A 1A 1A  1A 1A 1A 1A  1A 1A 1A 1A  
[   46.288452] RTW: 0x030: 00 00 00 EE  EE EE 00 00  00 EE 22 22  22 22 22 20  
[   46.288490] RTW: 0x040: 23 23 23 23  23 00 10 00  00 EE EE EE  26 26 26 26  
[   46.288528] RTW: 0x050: 28 29 2B 2D  30 30 2A 2C  2C 2C EE 10  11 EE EE FF  
[   46.288565] RTW: 0x060: 30 00 10 EE  21 21 21 21  21 1F 23 23  22 22 21 11  
[   46.288602] RTW: 0x070: 10 00 00 EE  EE EE 23 24  25 27 2A 2A  2A 2B 2B 2B  
[   46.288640] RTW: 0x080: 21 22 22 20  FF 30 01 EE  EE EE 20 10  00 EE 23 23  
[   46.288677] RTW: 0x090: 23 22 22 20  23 23 23 23  23 00 00 00  00 EE EE EE  
[   46.288715] RTW: 0x0A0: 27 27 28 28  21 21 22 22  22 22 1D 1E  1E 1E EE 11  
[   46.288752] RTW: 0x0B0: 10 EE EE EE  10 10 00 EE  7F 20 22 00  0F 00 00 88  
[   46.288789] RTW: 0x0C0: 88 01 00 00  00 00 00 55  00 FF 01 FF  FF FF FF FF  
[   46.288827] RTW: 0x0D0: DA 0B 13 88  E3 66 FF FF  00 C0 CA AC  87 AC 09 03  
[   46.288865] RTW: 0x0E0: 52 65 61 6C  74 65 6B 0E  03 38 30 32  2E 31 31 61  
[   46.288902] RTW: 0x0F0: 63 20 4E 49  43 08 03 31  32 33 34 35  36 FF FF FF  
[   46.288939] RTW: 0x100: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.288977] RTW: 0x110: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289014] RTW: 0x120: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF 04 FF FF  
[   46.289052] RTW: 0x130: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289089] RTW: 0x140: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289126] RTW: 0x150: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289164] RTW: 0x160: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289201] RTW: 0x170: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289238] RTW: 0x180: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289275] RTW: 0x190: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289313] RTW: 0x1A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289350] RTW: 0x1B0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289387] RTW: 0x1C0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289425] RTW: 0x1D0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289462] RTW: 0x1E0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289499] RTW: 0x1F0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF  
[   46.289540] RTW: hal_com_config_channel_plan chplan:0x7F
[   46.289565] RTW: [TRX_Nss] HALSPEC - tx_nss :3, rx_nss:3
[   46.289570] RTW: [TRX_Nss] Registry - tx_nss :0, rx_nss:0
[   46.289573] RTW: [TRX_Nss] HALDATA - tx_nss :2, rx_nss:3
[   46.290138] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[   46.290877] usb 1-6: USB disconnect, device number 14
`

I connects and recognized by the kernel but immediately disconnected. I physically reconnect it to the usb port and the cycle repeats. I'm using USB 3.0 cable and port on the back panel

Is there any fix?

morrownr, 2025-05-11:
If the port you are using is a USB3 port then, read the following:

https://github.com/morrownr/USB-WiFi/issues/508

If it is not a USB3 port, then do not set this parameter to 1.

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: added guard to prevent repeated forced USB mode-switch disconnect loops within a module lifetime. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.